mavlink/qgroundcontrol #6621: Display coordinates of RTK base station

mkunzBGC, 2018-06-21:
I am using QGroundControl 3.3.2 with APM:Copter 3.5.5 and a Here+ RTK base station and rover. While RTK works, there seems to be no way to find out the coordinates of the base station. This would be valuable information e.g. when performing surveys of the same area on different days!

As a first step, having just the coordinates displayed would help me a lot. According to Don Gagne, this should be quite easy to achieve:

> Don Gagne@DonLakeFlyer       Jun 20 19:34

>>    When using QGroundControl for RTK, is there a way to find out the coordinates of the base station?

>Add an Issue and I think this can be easily added to the GPSRTK FactGroup which would make it available to the instrument panel.

On the long run, it would be nice to be able to save and restore base station coordinates, similar to how MissionPlanner handles RTK ([link](http://ardupilot.org/copter/docs/common-here-plus-gps.html#base-module-setting-using-mission-planner)). 

Finally, for people like me who would also like to use RTK with QGC for ground-based surveying, a possibility to do copy and paste with base and rover positions would be wonderful (currently, rover coordinates are available via Mavlink Inspector, but I cannot copy from there).

Thanks for your great work,
Martin

hamishwillee, 2019-06-11:
@DonLakeFlyer You might want to close this

DonLakeFlyer, 2019-06-13:
> While RTK works, there seems to be no way to find out the coordinates of the base station. This would be valuable information e.g. when performing surveys of the same area on different days!

This statement is still correct. Those coordinates should be available for display in the instrument panel.

closed 2026-10-04 by github-actions[bot]