mavlink/qgroundcontrol #6624: Add support for ArduPilot's force-arm option

peterbarker, 2018-06-21:
ArduPilot has had a force-disarm option for quite some time.  Passing a "magic" value in as the second parameter of MAV_CMD_COMPONENT_ARM_DISARM would bypass any checks and instantly disarm the vehicle.  This is bared in MAVProxy as "disarm force", and in QGC as "Emergency Stop".

https://github.com/ArduPilot/ardupilot/pull/4891 has just been merged into ArduPilot.  It adds a "force arm" option which works in the same way.

You can read the reasoning for adding the option in that issue, but the most compelling reason to have this option is to stop people having to bypass arming checks by setting the ARMING_CHECK parameter to zero.  About half the crash logs I look at have that fault.  Several people have explicitly mentioned to me they were shocked when they realised they'd flown missions without having arming checks enabled when they'd earlier disabled them for testing purposes.

We are hoping that the GCS authors will add options to their interfaces to allow a user to force-arm their vehicles.  Obviously user-interfaces differ, but I was imagining that shift-clicking an "arming button" would be a force-arm, or perhaps a dwell-click followed by a confirmation on a tablet-style interface.  I don't know how sliders (as used by e.g. QGC) would be used to do a "force-arm".  (n.b. I'm not a UI designer ;-) )

This is the MAVProxy code which does the force-arm: https://github.com/ardupilot/MAVProxy/blob/master/MAVProxy/modules/mavproxy_arm.py#L94 - a user triggers it in MAVProxy with "arm throttle force".

DonLakeFlyer, 2018-06-21:
Thanks Peter.

DonLakeFlyer, 2018-06-28:
This should also go into the mavlink spec.

marcmerlin, 2019-07-15:
Thank you for this. The biggest need for me has been motor/amp testing at my desk in places where the GPS doesn't always allow arming.
Do you know which mavlink clients support this new force-arm? (mission planner, apmplanner, qgroundcontrol), and on android (tower, qgroundcontrol, other...)

marcmerlin, 2019-07-15:
just to cross-reference my own bug for the same request: https://github.com/ArduPilot/ardupilot/issues/3825

marcmerlin, 2019-07-16:
thanks, sounds like it may be the only client that support this for now.
But one is good enough, I'll switch to it :)

DonLakeFlyer, 2019-07-16:
@peterbarker I don't want to add this as something ArduPilot specific. Can you get an update to mavlink spec through for this then I'll implement?

marcmerlin, 2019-07-16:
actually @peterbarker , is there a single android client that supports this?

peterbarker, 2019-07-16:
On Tue, 16 Jul 2019, Marc MERLIN wrote:

> actually @peterbarker , is there a single android client that supports 
> this?

It is *possible* Solex does - but I doubt any other does.

karambolis8, 2019-09-30:
This would be great to have it. Very useful feature when testing new airframes in no-GPS places like basement.

DonLakeFlyer, 2019-09-30:
@peterbarker Another ping to get this to be mavlink standard?

DonLakeFlyer, 2019-09-30:
@hamishwillee This would seem to make sense as a standard thing to me.

hamishwillee, 2019-10-01:
@DonLakeFlyer It already is - see [MAV_CMD_COMPONENT_ARM_DISARM](https://mavlink.io/en/messages/common.html#MAV_CMD_COMPONENT_ARM_DISARM).

I'm pretty sure PX4 supports this too

DonLakeFlyer, 2019-10-01:
"0: only arm-disarm when landed, 21196: enforce arming/disarming even in-air (during flight)"

@hamishwillee That's not quite what this is. This is a "force arm" which skips all pre-arm checks. It's not really anything about being in-flight.

peterbarker, 2019-10-01:
On Tue, 1 Oct 2019, Don Gagne wrote:

> "0: only arm-disarm when landed, 21196: enforce arming/disarming even in-air (during flight)"

Not sure where that comes from.

     static constexpr const float magic_force_arm_value = 2989.0f;
     static constexpr const float magic_force_disarm_value = 21196.0f;

https://github.com/ArduPilot/ardupilot/blob/master/libraries/GCS_MAVLink/GCS.h#L435

DonLakeFlyer, 2019-10-01:
> Not sure where that comes from.

Current mavlink spec. 

That said the concept of requiring a magic number to arm in the air sounds odd to me though. Seems like arming in the air should work by default. Certainly QGC has not concept of supporting a special "arm in the air" mode of arm.

hamishwillee, 2019-10-01:
@DonLakeFlyer I think the position of the open bracket is making people confused (even though it is not actually ambiguous). So magic number is just force the arm or disarm - even if flying or not passing pre-flight checks.

> 0: only arm-disarm when landed, 21196: enforce arming/disarming (even in-air during flight).

OR slightly improved

> 0: only arm-disarm when landed, 21196: enforce arming/disarming (even when not passing prearm checks or during flight).

Is there a reason to have a separate number for forcing arm vs disarm? If not, then perhaps ArduPilot would be best to just use the same number for both.

DonLakeFlyer, 2019-10-01:
> I think the position of the open bracket is making people confused

Ah, I see now!

> Is there a reason to have a separate number for forcing arm vs disarm?

Not that I can think of.

hamishwillee, 2019-10-02:
@DonLakeFlyer That was my interpretation. It is equally possible that only the in-flight disarm case was considered.

I have created PR to "test" this: https://github.com/mavlink/mavlink/pull/1243

peterbarker, 2019-10-02:
On Tue, 1 Oct 2019, Don Gagne wrote:

> That said the concept of requiring a magic number to arm in the air sounds odd to me though. Seems like arming in the air should work by default.

The really big problem there is knowing when you're flying.  That's 
actually really quite difficult - especially if you don't have an airspeed 
sensor.

Easiest way to think of that is somebody standing in a 20 knot wind 
holding their plane in the wind - is that vehicle flying or not?  How 
about when the pilot simply lets go of the aircraft and it sits in mid-air 
- now it's definitely flying, but ArduPilot doesn't currently support a 
"grip" sensor ;-)

> Certainly QGC has not concept of supporting a special "arm in the air" mode of arm.

Well, neither does ArduPilot - just force-arm.  (Actually, bit of a 
simplification, there's a corner case which you could consider "arm in the 
air" - when we detect a critical failure we may handle arming specially).

DonLakeFlyer, 2019-10-02:
@peterbarker The in the air thing is just bad wording. The current mavlink spec is supposed to mean that the magic number for force disarm can also be used for force arm (at any time). Given that, was there a reason ArduPilot chose a different magic number for force arm?

peterbarker, 2019-10-03:
On Wed, 2 Oct 2019, Don Gagne wrote:

> @peterbarker The in the air thing is just bad wording. The current mavlink spec is supposed to mean that the magic number for force disarm can also be
> used for force arm (at any time). Given that, was there a reason ArduPilot chose a different magic number for force arm?

No, but it's been that way for over 2.5 years :-)

DonLakeFlyer, 2019-10-03:
Ok, so PX4 is implemented such that 21196 only works for forcing a disarm. It does nothing with that value for arm. Hence force arm could theoretically be changed to 2989 without causing problems with at least well known firmwares.

hamishwillee, 2019-10-03:
@peterbarker @DonLakeFlyer Yes it could and we will if necessary based on it being "prior art". But to me using the same value is a cleaner story, so my _preference_ would be that ArduPilot accept either value for the force arm - breaks nothing, one line change.

DonLakeFlyer, 2019-10-04:
> But to me using the same value is a cleaner story

I would tend to agree with that.

peterbarker, 2019-10-04:
I've asked Craig to put it on the agenda for next week's devcall.

Pr which added force-arm for ArduPilot is here: https://github.com/ArduPilot/ardupilot/pull/4891

hamishwillee, 2019-10-06:
Thanks @peterbarker - if this comes down to "we don't care but no one has time to do it" then I'm happy to make the change in ArduPilot code myself.

peterbarker, 2019-10-06:
On Sun, 6 Oct 2019, Hamish Willee wrote:

> Thanks @peterbarker - if this comes down to "we don't care but no one has time to do it" then I'm happy to make the change in ArduPilot code myself.

It's a simple change, but we would have to carry it forever.

I'd also like to point out that being a bit thorny in this area is good 
IMO.  Having the GCS author consider the separate cases when implementing 
is good - a forced disarm is nowhere near as dangerous as a force-arm 
(generally).

hamishwillee, 2019-10-06:
@peterbarker 

I'm not _particularly_ convinced by that argument - they are both very cases that both deserve serious consideration. But we're already using a magic number for this rather than a boolean, so it is always going to be a deliberate choice.

tridge, 2019-10-07:
>  Given that, was there a reason ArduPilot chose a different magic number for force arm?

It was in ArduPilot first. Whoever brought it across from ArduPilot to the xml didn't bring it over properly

hamishwillee, 2019-10-08:
@tridge That wasn't the question. I accept it is prior art, but is it _better_ than having a single value?

closed 2020-08-15 by DonLakeFlyer