mavlink/qgroundcontrol #6626: Comm errors downloading DF logs over USB on Windows10

Kelly-Foster, 2018-06-21:
Installed QGC today (21jun18) on Windows 10.  Bench testing fresh build from ArduPlane master this morning.  Matek F405-Wing flight controller with Spedix 35A BLHeli32 ESC, generic motor, GPS, digital airspeed sensor (on I2C), 3 flight control servos, RunCam Split 2, Dragonlink NanoRx talking SBus in to FC.  Ran a few brief sessions bench testing, and in exploring QGC, tried downloading the last few DF logs.  Frequent "Communications Lost" voice alerts, and screen eventually shows one log TIMED OUT, and one log "Error".  Yet both are in my log directory, and I could browse them in MissionPlanner (as a double check on logs created w/o MP).  

First screen grab shows the QGC version in upper left corner, and yellow error popup dialog when comm errors occur.  Second screen grab shows log status window after download.

Kelly

[image]

[image]

DonLakeFlyer, 2018-06-22:
@acfloria Can you look at this. I think it was broken by all the heartbeat and communication lost changes. LogDownload uses Vehicle::setConnectionLostEnabled to turn off communication lost errors while downloading. But it doesn't seem to work any more.

acfloria, 2018-06-25:
Ok, I will have a look at it tomorrow.

DonLakeFlyer, 2018-06-26:
This also shows the comm lost due to no heartbeats only problem. The link is fine. It's just full of download packets, choking out heartbeats.

DonLakeFlyer, 2018-06-26:
This is fixed now

closed 2018-06-26 by DonLakeFlyer