mavlink/qgroundcontrol #6629: Wrong default altitude in new altitude selector

okalachev, 2018-06-21:
In the `master` branch version of QGC the altitude of a Land waypoint defaults to the flight altitude: 

[image]

But [MAVLink specifies](http://mavlink.org/messages/common#MAV_CMD_NAV_LAND) that this is the ground altitude, so I think, that it should default to 0, as the 3.3.2 version does:

[image]

DonLakeFlyer, 2018-06-23:
This UI isn't updating correctly. The actual value send to vehicle is 0.

DonLakeFlyer, 2018-06-23:
Both _LAND and _VTOL_AND are forced back to Relative:0.

DonLakeFlyer, 2018-06-23:
Thanks for the bug report.

closed 2018-06-23 by DonLakeFlyer