mavlink/qgroundcontrol #6631: Minor suggestions for improved UI and functionality

Kelly-Foster, 2018-06-21:
**QGC:**  Installed QGC today (21jun18) on Windows 10. Bench testing fresh build from ArduPlane master this morning.  **Hardware:**  Matek F405-Wing flight controller with Spedix 35A BLHeli32 ESC, generic motor, GPS, digital airspeed sensor (on I2C), 3 flight control servos, RunCam Split 2, Dragonlink NanoRx talking SBus in to FC.  RxRSSI via PWM values on Ch12.

Minor suggestions for improvement, relative to years of using MissionPlanner:
•	Top request is compare params facility.  This is extremely useful when FC has been flashed with new FW, and flashing overwrites with default params.  Param compare with selective checkbox per param is really useful.
•	Mavlink Inspector initial window is quite small on my screen.  Can it be sized to be more like 50% of width by 50% of height of screen?  Or % of QGC main window?
•	Mavlink Inspector is a great tool.  Really like it.  I noticed it stays up and eventually occludes windows like Arm/Disarm, RxRSSI, etc.  Is it possible for most recent action's window to always be uppermost?
•	Can’t select text at top of QGC window when wanting to reference a particular build hash.  Any way to make that text selectable?
•	How to get QGC upgrades, ala MP?  I can't tell if I am on Stable, Beta, or latest daily?
•	Is there any way to reboot FC from QGC?

Thanks.  Really impressed with QGC overall.

Kelly

DonLakeFlyer, 2018-06-22:
> Haven't been able to get firmware flashing screen to work at all.

I assume from @lvale comment you are using Chibios boards/firmware. Flashing for that is not yet supported.

> Is there any way to save my last connection (COM21 at 115k baud) and next time check that port, that speed, and if you think there is an Ardupilot vehicle connected, go ahead and autoconnect to it.

You can create a comm link from settings and you can set it to autoconnect.

DonLakeFlyer, 2018-06-25:
> How to get QGC upgrades, ala MP? I can't tell if I am on Stable, Beta, or latest daily?

Only Android builds auto-update. The rest you have to manually update.

DonLakeFlyer, 2019-12-24:
Mavlink Inspector was completely redone. All new in daily builds.

DonLakeFlyer, 2019-12-24:
> Is there any way to save my last connection (COM21 at 115k baud) and next time check that port, that speed, and if you think there is an Ardupilot vehicle connected, go ahead and autoconnect to it.

ChibiOS autoconnect should be all good in daily builds. Also under Setup/Comm Links you can also create your own manual links and set them to AutoConnect if you want. But you need to turn off regular AutoConnect from settigns for that so they don't conflict.

DonLakeFlyer, 2019-12-24:
Moving Compare Params to it's own issue. The rest should be taken care of

closed 2019-12-24 by DonLakeFlyer