mavlink/qgroundcontrol #6650: "Values" panel freezes, position on map not updated

mkunzBGC, 2018-06-25:
I am using QGroundControl 3.3.2 on Ubuntu 18.10 with APM:Copter 3.5.5 and a Here+ RTK base station and rover.

Today I carried out some test flights using RTK. It happened two times that a part of the "Values" panel froze: artificial horizon, compass and Ground Speed value were still updated, but other values (Altitude-rel, Altitude, Temperature, Voltage) just stayed constant. At the same time, the position of the vehicle on the map froze: The red arrow stopped moving while still turning in sync with the copter.

[Here is a tlog that shows the problem](https://oc.bgc-jena.mpg.de/index.php/s/5aUHH9Zbxs3dPBH). The freezing happens at about 2/3 of the total log time.
[Here is the dataflash log belonging to the tlog.](https://oc.bgc-jena.mpg.de/index.php/s/uyKhxAtXEEMBTij)

DonLakeFlyer, 2018-06-25:
You're going to need a firmware person to look at this. QGC is updating things because it is nl onger getting mavlink messages for vehicle position. You can see this at around the 11 minute mark when you replay the telemetry log through a daily build (daily builds have lots of log replay fixes in it) and turn on mavlink insepector.

mkunzBGC, 2018-07-03:
Thanks for looking into this, @DonLakeFlyer . How do we get a firmware person? Shall I open an issue on the ArduPilot bug tracker? If so, rather in the QCS or the Copter 3.5 category?

DonLakeFlyer, 2018-07-03:
I'd start by posting the problem in an appropriate ArduPilot forum: https://discuss.ardupilot.org. And then see where that leads.

mkunzBGC, 2018-07-16:
I've posted it [here](https://discuss.ardupilot.org/t/copter-stops-sending-vehicle-position-to-gcs/30571), but unfortunately got no response so far.

closed 2026-10-04 by github-actions[bot]