mavlink/qgroundcontrol #6657: Multiple vehicle connection make to be turned off.

Seunghwan17, 2018-06-26:
Hello.
I have some bug about multiple vehicle.
I have more than 20 drones. So I connect drones to QGC. When I connected, 13 drones was connected with QGC. But I try to connect over the 13drones, then QGC was crushed without any message or error.
Anybody have this experience ?

DonLakeFlyer, 2018-06-26:
What OS? Are you using a Daily BUild?

Seunghwan17, 2018-06-27:
OS is windows and build is v3.3.2.
maybe I think that version is latest version at that time.

DonLakeFlyer, 2018-06-27:
First thing to dry is switch to a Daily Build.

Seunghwan17, 2018-06-28:
Daily build is recently version? or what is mean?

DonLakeFlyer, 2018-06-28:
https://docs.qgroundcontrol.com/en/releases/daily_builds.html

Daily builds are the newest QGC in development.

closed 2026-10-04 by github-actions[bot]