mavlink/qgroundcontrol #6661: PreFlight checklist lockdown for Stable

DonLakeFlyer, 2018-06-26:
This needs to be cleaned up. Current state is more of a fixed wing list.

philipoe, 2018-06-28:
@DonLakeFlyer I assume you intend to make this configurable per vehicle, and _not_ make it so generic that it fits every kind of vehicle (which would be quite useless, i guess you agree with that)?

DonLakeFlyer, 2018-06-28:
Yup. I haven't put much thought into that though. Have you?

philipoe, 2018-06-28:
Not yet, no. I need to look into the new checklist structure (given the quite extensive changes you did) first in any case, but am gone for a project demonstration for the next ~3 weeks.

Antiheavy, 2020-02-15:
@DonLakeFlyer this issue seems overcome by events?

DonLakeFlyer, 2020-02-15:
I think what @dogmaphobic is good enough for now

closed 2020-02-15 by DonLakeFlyer