mavlink/qgroundcontrol #6675: QML Module Not found  In Qt 5.11.0.

MyOnlyOnlySunshine, 2018-07-02:
Hello, when I use qt5.11.0 to open the source code compiler, the compiler reported an error, but can successfully compile, but can not use those who complained qml, also ask why the source of the source will be more Come out three temporary compilation files, what is that? Thank you

[image]
[image]

DonLakeFlyer, 2018-07-02:
No quite sure what you are asking. But you cannot edit QGC qml files in the designer. They can only be edited using the text editor. Those warnings you show for module not found can be ignored.

closed 2018-07-02 by DonLakeFlyer

LimEunji, 2018-09-27:
Why can't I edit QGC qml files in the designer?