mavlink/qgroundcontrol #6679: Add NXPHlite V3

davids5, 2018-07-03:
@DonLakeFlyer 

Would you please add the USB auto connect for PX4 NXPHlite v3.x

CONFIG_CDCACM_VENDORID=0x1FC9
CONFIG_CDCACM_VENDORSTR="NXP"
 CONFIG_CDCACM_PRODUCTSTR="PX4 NXPHlite v3.x"
CONFIG_CDCACM_PRODUCTID=0x001c

The firmware will be under development for a bit but will be in nxphlite-v3_default.px4 when released.

davids5, 2018-07-04:
I should have also mentioned the board_id is 28 decimal (0x1c). It is the same as the product id.

davids5, 2018-07-06:
@dogmaphobic would you be able to add this?

dogmaphobic, 2018-07-06:
> @dogmaphobic would you be able to add this?

Sure! Blindly but you can let me know if it works...

DonLakeFlyer, 2018-07-06:
What is the board id from the bootloader for this board?

davids5, 2018-07-06:
board_id is 28 decimal (0x1c). It is the same as the product id

LorenzMeier, 2018-07-12:
A pull request with those extensions would be most appreciated. It has probably reached the point where board support should be factored out of the code and into an XML file or equivalent. I'm closing this as it is not a real issue and something that could be easily contributed. Thanks for submitting the PR and I'm closing this here.

closed 2018-07-12 by LorenzMeier