mavlink/qgroundcontrol #6681: SSL error on Appimage

kikislater, 2018-07-04:
- Ground station operating system : Archlinux x64

- QGroundControl version and build : latest daily build Appimage : aeae358a2 2018-07-03 09:31:52 -0700

- Autopilot board: not required

- Autopilot firmware : not required.

- Exact steps to reproduce the problem : 
Plan survey mission, add terrain

Log file below : 

> [D] at /home/travis/build/mavlink/qgroundcontrol/src/QGCApplication.cc:257 - "Settings location "/home/kiki/.config/QGroundControl.org/QGroundControl.ini" Is writable?: true"
> [D] at /home/travis/build/mavlink/qgroundcontrol/src/QGCLoggingCategory.cc:97 - "Filter rules "*Log.debug=false\nTerrainTileLog.debug=true\nTerrainQueryLog.debug=true\nTerrainQueryVerboseLog.debug=true\n""
> [D] at /home/travis/build/mavlink/qgroundcontrol/src/PositionManager/PositionManager.cpp:39 - "QGeoPositionInfoSourceGeoclueMaster(0x1d474e0)"
> [!] at :0 - "Failed to set Geoclue positioning requirements. Geoclue error: org.qtproject.QtDBus.Error.InvalidObjectPath"
> [E] at /home/travis/build/mavlink/qgroundcontrol/src/PositionManager/PositionManager.cpp:111 - "QGCPositionManager error 2"
> [D] at /home/travis/build/mavlink/qgroundcontrol/src/Vehicle/MAVLinkLogManager.cc:360 - "MAVLink logs directory: "/home/kiki/Documents/QGroundControl/Logs""
> [D] at /home/travis/build/mavlink/qgroundcontrol/src/QtLocationPlugin/QGCMapEngine.cpp:225 - "Map Cache in: "/home/kiki/.cache/QGCMapCache300" / "qgcMapCache.db""
> [D] at /home/travis/build/mavlink/qgroundcontrol/src/comm/UDPLink.cc:233 - "Adding target QHostAddress("127.0.0.1") 14550"
> [D] at /home/travis/build/mavlink/qgroundcontrol/src/comm/MAVLinkProtocol.cc:436 - "Temp log "/tmp/FlightData680327.mavlink""
> [D] at /home/travis/build/mavlink/qgroundcontrol/src/comm/UDPLink.cc:233 - "Adding target QHostAddress("127.0.0.1") 14550"
> [E] at :0 - "QSslSocket: cannot resolve CRYPTO_num_locks"
> [E] at :0 - "QSslSocket: cannot resolve CRYPTO_set_id_callback"
> [E] at :0 - "QSslSocket: cannot resolve CRYPTO_set_locking_callback"
> [E] at :0 - "QSslSocket: cannot resolve ERR_free_strings"
> [E] at :0 - "QSslSocket: cannot resolve EVP_CIPHER_CTX_cleanup"
> [E] at :0 - "QSslSocket: cannot resolve EVP_CIPHER_CTX_init"
> [E] at :0 - "QSslSocket: cannot resolve sk_new_null"
> [E] at :0 - "QSslSocket: cannot resolve sk_push"
> [E] at :0 - "QSslSocket: cannot resolve sk_free"
> [E] at :0 - "QSslSocket: cannot resolve sk_num"
> [E] at :0 - "QSslSocket: cannot resolve sk_pop_free"
> [E] at :0 - "QSslSocket: cannot resolve sk_value"
> [E] at :0 - "QSslSocket: cannot resolve SSL_library_init"
> [E] at :0 - "QSslSocket: cannot resolve SSL_load_error_strings"
> [E] at :0 - "QSslSocket: cannot resolve SSL_get_ex_new_index"
> [E] at :0 - "QSslSocket: cannot resolve SSLv3_client_method"
> [E] at :0 - "QSslSocket: cannot resolve SSLv23_client_method"
> [E] at :0 - "QSslSocket: cannot resolve SSLv3_server_method"
> [E] at :0 - "QSslSocket: cannot resolve SSLv23_server_method"
> [E] at :0 - "QSslSocket: cannot resolve X509_STORE_CTX_get_chain"
> [E] at :0 - "QSslSocket: cannot resolve OPENSSL_add_all_algorithms_noconf"
> [E] at :0 - "QSslSocket: cannot resolve OPENSSL_add_all_algorithms_conf"
> [E] at :0 - "QSslSocket: cannot resolve SSLeay"
> [E] at :0 - "Incompatible version of OpenSSL"
> [D] at /home/travis/build/mavlink/qgroundcontrol/src/Terrain/TerrainQuery.cc:275 - "supportsSsl false sslLibraryBuildVersionString "OpenSSL 1.0.2k-fips  26 Jan 2017""
> [D] at /home/travis/build/mavlink/qgroundcontrol/src/Terrain/TerrainQuery.cc:524 - "TerrainAtCoordinateBatchManager::_sendNextBatch _state:_requestQueue.count:_sentRequests.count "Idle" 2 0"
> [D] at /home/travis/build/mavlink/qgroundcontrol/src/Terrain/TerrainQuery.cc:552 - "_sendNextBatch - batch count:request queue count 2 0"
> [D] at /home/travis/build/mavlink/qgroundcontrol/src/Terrain/TerrainQuery.cc:338 - "TerrainTileManager::addCoordinateQuery count 2"
> [D] at /home/travis/build/mavlink/qgroundcontrol/src/Terrain/TerrainQuery.cc:497 - "Computing unique tile hash for  QGeoCoordinate(-20.201236737, 57.534704082, 0) "80000002375300006979001""
> [D] at /home/travis/build/mavlink/qgroundcontrol/src/Terrain/TerrainQuery.cc:390 - "Need to download tile  "80000002375300006979001""
> [E] at :0 - "QSslSocket: cannot call unresolved function SSLv23_client_method"
> [E] at :0 - "QSslSocket: cannot call unresolved function SSL_CTX_new"
> [E] at :0 - "QSslSocket: cannot call unresolved function SSL_library_init"
> [E] at :0 - "QSslSocket: cannot call unresolved function ERR_get_error"
> [E] at /home/travis/build/mavlink/qgroundcontrol/src/Terrain/TerrainQuery.cc:450 - "Elevation tile fetching returned error ( QNetworkReply::NetworkError(UnknownNetworkError) )"
> [D] at /home/travis/build/mavlink/qgroundcontrol/src/Terrain/TerrainQuery.cc:613 - "TerrainAtCoordinateBatchManager::_coordinateHeights success:count false 0"
> [E] at :0 - "QSslSocket: cannot call unresolved function ERR_get_error"
> [D] at /home/travis/build/mavlink/qgroundcontrol/src/Terrain/TerrainQuery.cc:275 - "supportsSsl false sslLibraryBuildVersionString "OpenSSL 1.0.2k-fips  26 Jan 2017""
> [D] at /home/travis/build/mavlink/qgroundcontrol/src/Terrain/TerrainQuery.cc:697 - "TerrainPolyPathQuery::requestData count 188"
> [D] at /home/travis/build/mavlink/qgroundcontrol/src/Terrain/TerrainQuery.cc:371 - "TerrainTileManager::addPathQuery start:end:coordCount QGeoCoordinate(-20.196878676, 57.522613519) QGeoCoordinate(-20.197148472, 57.522613519) 3"
> [D] at /home/travis/build/mavlink/qgroundcontrol/src/Terrain/TerrainQuery.cc:497 - "Computing unique tile hash for  QGeoCoordinate(-20.196878676, 57.522613519) "80000002375200006980001""
> [D] at /home/travis/build/mavlink/qgroundcontrol/src/Terrain/TerrainQuery.cc:390 - "Need to download tile  "80000002375200006980001""
> [D] at /home/travis/build/mavlink/qgroundcontrol/src/Terrain/TerrainQuery.cc:524 - "TerrainAtCoordinateBatchManager::_sendNextBatch _state:_requestQueue.count:_sentRequests.count "Idle" 1 0"
> [D] at /home/travis/build/mavlink/qgroundcontrol/src/Terrain/TerrainQuery.cc:552 - "_sendNextBatch - batch count:request queue count 1 0"
> [D] at /home/travis/build/mavlink/qgroundcontrol/src/Terrain/TerrainQuery.cc:338 - "TerrainTileManager::addCoordinateQuery count 1"
> [D] at /home/travis/build/mavlink/qgroundcontrol/src/Terrain/TerrainQuery.cc:497 - "Computing unique tile hash for  QGeoCoordinate(-20.196878676, 57.522613519) "80000002375200006980001""
> [D] at /home/travis/build/mavlink/qgroundcontrol/src/Terrain/TerrainQuery.cc:390 - "Need to download tile  "80000002375200006980001""
> [E] at :0 - "QSslSocket: cannot call unresolved function SSLv23_client_method"
> [E] at :0 - "QSslSocket: cannot call unresolved function SSL_CTX_new"
> [E] at :0 - "QSslSocket: cannot call unresolved function SSL_library_init"
> [E] at :0 - "QSslSocket: cannot call unresolved function ERR_get_error"
> [E] at /home/travis/build/mavlink/qgroundcontrol/src/Terrain/TerrainQuery.cc:450 - "Elevation tile fetching returned error ( QNetworkReply::NetworkError(UnknownNetworkError) )"
> [D] at /home/travis/build/mavlink/qgroundcontrol/src/Terrain/TerrainQuery.cc:707 - "TerrainPolyPathQuery::_terrainDataReceived success:_curIndex false 0"
> [D] at /home/travis/build/mavlink/qgroundcontrol/src/Terrain/TerrainQuery.cc:613 - "TerrainAtCoordinateBatchManager::_coordinateHeights success:count false 0"
> [E] at :0 - "QSslSocket: cannot call unresolved function ERR_get_error"

DonLakeFlyer, 2018-07-05:
I believe this is because ArchLinux doesn't have installed by default the version of OpenSSL that Qt works with. You'll have to do some digging around in Qt land to figure out which one that is. I believe some other folks may have already figured out what that is. Whereas this is all working fine on Ubunutu which is the AppImage tested version.

closed 2018-07-05 by DonLakeFlyer

kikislater, 2018-07-12:
SSL is installed and working in my qt5 installation !

$ ldd /usr/lib/libQt5Network.so | grep libssl
	libssl.so.1.1 => /usr/lib/libssl.so.1.1 (0x00007f39c3044000)

Which version of ssl is required in app image ?

kikislater, 2018-07-12:
I built it myself and it's ok, I got terrain working. But still have this ssl error on app image ! 
Easy to build, solve for me but may be not for others ...

DonLakeFlyer, 2018-07-13:
I think Qt wants 1.0 as oppposed to 1.1.