mavlink/qgroundcontrol #6683: Comm Links buggy UI

alchemyconsulting, 2018-07-05:
----
When posting bug reports, include the following informaiton:
- Mac OSX, Android 8.0
- QGC HEAD:2018-06-22, downloaded from Dailies (but seen same behaviour from stable releases)

When you add a new connection under Comm Links, a few bugs:
 1. If you type a name in, and then change the type, the name resets to Unnamed
 2. If you create new UDP connection, and add a Target host and specify a hostname, it gets translated to IP address.  It should keep hostname, so if IP changes through dhcp it follows to new IP.
 3. If you create new UDP connection, add a Target Host, it's not obvious/intuitive that you have to specify host:port syntax.  Would be nice to have specific boxes for host/ip and port, labelled.
 4. If you create new UDP connection, add a Target Host, and then click OK to save the connection, if you go back to edit the Connection usually the target host has disappeared.  It does seem to still exist as it sometimes connects, but it doesn't show up.  Very confusing.  This is the biggest problem.

DonLakeFlyer, 2018-07-05:
> If you create new UDP connection, add a Target Host, and then click OK to save the connection, if you go back to edit the Connection usually the target host has disappeared. It does seem to still exist as it sometimes connects, but it doesn't show up. Very confusing. This is the biggest problem.

Fixed in daily builds.

alchemyconsulting, 2018-07-06:
Hi, I've just downloaded and tried with last night's daily build (2018-07-05 22:15:03) and still have the same problem?

DonLakeFlyer, 2018-07-06:
Sorry, just realized it hasn't been merged yet: https://github.com/mavlink/qgroundcontrol/pull/6672.

DonLakeFlyer, 2018-07-06:
Just merged. try an OSX daily from tomorrow. Android dailies are still not updating.

closed 2018-07-06 by DonLakeFlyer