mavlink/qgroundcontrol #6685: Segmentation fault if exit from offline maps view before map loaded

Rogni, 2018-07-05:
1. Start QGroundControll
2. Go to Application settings view
3. Go to Offline Maps
4. Click on "Add New Set"
5. Change map zoom and click on "General" button before map loaded (gray screen instead of map)

result: Segmentation fault (core dumped)

Qt 5.11.0
gcc (Ubuntu 7.3.0-16ubuntu3) 7.3.0
g++ (Ubuntu 7.3.0-16ubuntu3) 7.3.0

 OS: Ubuntu 18.04 bionic
 Kernel: x86_64 Linux 4.15.0-24-generic
 Resolution: 1366x768
 DE: MATE 1.20.0
 CPU: Intel Pentium 2117U @ 2x 1.8GHz [55.0°C]
 GPU: GeForce GT 720M

DonLakeFlyer, 2019-01-19:
I believe this was the terrain crash

DonLakeFlyer, 2019-01-19:
@Rogni You can reopen if you can still make it happen on latest daily builds

closed 2019-01-19 by DonLakeFlyer