mavlink/qgroundcontrol #6693: ArduPilot: Add support for flashing Chibios firmwares

DonLakeFlyer, 2018-07-07:
From Tridge:

"I'd like to discuss how the new ChibiOS based boards are detected and
correct firmware chosen from the firmware.ardupilot.org website.

Up to now we've used the board IDs from the bootloader, but that didn't
give a good way to know the board type when the bootloader is not
running.

For the ChibiOS builds we've made the USB IDs contain the firmware type
in the USB strings. So for fmuv5 I see this on Linux:

 /dev/serial/by-id/usb-ArduPilot_fmuv5_2A001E000151373137393536-if00

where "fmuv5" is the build type.

On windows it shows the "Bus reported device description" as fmuv5.

 http://uav.tridgell.net/tmp/fmuv5.png"

DonLakeFlyer, 2018-07-07:
@tridge So for boards which have not been flashed yet. Should the Chibios builds be allowed to flash onto boards which report the bootloader boardid as:

```    static const int boardIDPX4FMUV5 = 50;      ///< PX4 V5 board, as from USB PID```

DonLakeFlyer, 2018-07-07:
And then in that case the user has two firmware options?
* Chibios version
* Old NuttX version

Or should it only allow Chibios?

Hwurzburg, 2018-07-08:
This may or may not be related, but Qgroundcontrol does not connect to my OmnibusF4pro using ChiBios firmware..neither does Tower....Tower does recognize the F4light HAL firmware on the board...Qgroundcontrol did not...

LorenzMeier, 2018-07-12:
@rmackay9 You can continue the discussion here.

DonLakeFlyer, 2018-07-13:
> This may or may not be related, but Qgroundcontrol does not connect to my OmnibusF4pro using ChiBios firmware..neither does Tower....Tower does recognize the F4light HAL firmware on the board...Qgroundcontrol did not...

You need to manual connect to the board using a manual connection. It will not autoconnect

Hwurzburg, 2018-07-13:
This is an Android USB connection,not PC...how do you manually connect?...the USB port is not listed as a comm port when in Android looking under manual connections

DonLakeFlyer, 2019-01-19:
This is implemented now

Hwurzburg, 2019-01-19:
MUCH APPRECIATED! just tested on Omnibus and a WingFC10....thanks!

DonLakeFlyer, 2019-01-20:
QGC should connect if you are running ArduPilot ChibiOS firmware on it.

DonLakeFlyer, 2019-01-20:
Are you running a daily build?

Hwurzburg, 2019-01-20:
yes.....1/18

DonLakeFlyer, 2019-01-20:
Ok, let me figure out where my test board has gone to and I'll take a look.

DonLakeFlyer, 2019-01-20:
ChibiOS working now. DragonLink still not working but that will be a separate issue.

closed 2019-01-20 by DonLakeFlyer

DonLakeFlyer, 2019-01-21:
DragonLink is covered in #7175