mavlink/qgroundcontrol #6701: Support uploading ArduPilot to non-Pixhawk boards

rmackay9, 2018-07-10:
ArduPilot is being ported to a growing number of smaller flight-controllers thanks to ChibiOS.

I've created a [wiki page that explains the easiest method to get ArduPilot onto these boards](http://ardupilot.org/copter/docs/common-loading-firmware-onto-chibios-only-boards.html) (the first time) but it involves using the Betaflight Configurator.  It's a fine tool but it would be nice if the Install Firmware screen would allow uploading ArduPilot using dfu.  The firmware can also be loaded using dfu-utils and the [instructions for that method are here](http://ardupilot.org/copter/docs/common-openpilot-revo-mini.html).

I'm not sure of the best way this functionality can be integrated into QGC of course so perhaps it requires some discussion.  The core issue is that the boards need to have an ArduPilot compatible bootloader loaded onto them.  Somewhere on firmware.ardupilot.org we have both stand-alone bootloaders for the boards but we also have special versions of ArduPilot that included the bootloader (i.e. flight code + bootloader) so either will work.

LorenzMeier, 2018-07-12:
A pull request with those extensions would be most appreciated. It has probably reached the point where board support should be factored out of the code and into an XML file or equivalent. I'm closing this as it is not a real issue and something that could be easily contributed. Looking forward to the PR!

DonLakeFlyer, 2018-07-13:
I'm going to use this one as the feature request since it has all the details. Not sure when it will happen since it requires a completely different flash procedure.

DonLakeFlyer, 2018-07-13:
#6693 is a bit different since that is for flashing to Pixhawk style boards which should be easy. I'll update that title for that one.

closed 2026-10-04 by github-actions[bot]