mavlink/qgroundcontrol #6705: Add Sentera Single NDVI sensor to camera list

J1100, 2018-07-11:
Please add the following camera and settings to QGC as an option:

Sentera NDVI Single Sensor

Sensor Width mm: 4.68001
Sensor Height mm: 3.56251
Image width px: 1248
Image height px: 952
Focal length mm: 4.14mm

Thank you!

J1100, 2018-07-11:
Also please add Sentera Dual Sensor:

Sensor Width (actual sensor in the camera NOT the size of the camera)  in mm: 6.2
Sensor Height (actual sensor in the camera NOT the size of the camera)  in mm: 4.65
Image width px: 4000
Image height px: 3000
Focal length mm: 5.4

Antiheavy, 2018-07-14:
+1 for this request

DonLakeFlyer, 2018-07-30:
@Antiheavy What are the official names for these two?

Antiheavy, 2018-07-30:
Sentera NDVI Single Sensor
Sentera Double 4K Sensor

The specs above are correct, or close enough anyway.