mavlink/qgroundcontrol #6707: Create new official App Store Stable Version

J1100, 2018-07-11:
Daily builds are changing rapidly and the only official release is 3.2 which is full of bugs and lacking features. It would be very much appreciated if QGC could get a new official App Store release up and running for us to point customers to. We would love to use QGC as our default app when selling new drones as well but this is the primary issue holding us back.
Thank you!

DonLakeFlyer, 2018-07-13:
I assume you are talking about iOS?

J1100, 2018-07-16:
Android, actually. I see the version up is 3.3.2 not 3.2 as I stated (3.2 release notes are the latest shown) but my issues still remain, it's not very reliable and is missing features of the newer builds. Thanks

DonLakeFlyer, 2018-07-16:
3.4 Stable will be out in a few weeks.

closed 2018-07-16 by DonLakeFlyer