mavlink/qgroundcontrol #6709: Does not recognize or connect to ChiBios Arudpilot boards like OmnibusF4Pro

Hwurzburg, 2018-07-11:
Cannot use QGC with OmnibuF4PRo running ChiBios based Ardupilot.....will not connect either in Adroid nor Windows version...

Hwurzburg, 2018-07-12:
I was mistaken....in Windows the USB serial port can be added and connection made...in Adroid, even though permission is given to QGC to use the USB, when a SERIAL connection is added, there is no port found...

LorenzMeier, 2018-07-12:
A pull request with those extensions would be most appreciated. It has probably reached the point where board support should be factored out of the code and into an XML file or equivalent. I'm closing this as it is not a real issue and something that could be easily contributed. Looking forward to the PR!

closed 2018-07-12 by LorenzMeier

Hwurzburg, 2018-07-12:
Sorry, not something I am capable of producing....not a programmer

Hwurzburg, 2019-01-20:
More and more users are having an issue with this....it is a real issue and should not be closed dismissively expecting a non-programmer/ non-dev to resolve it....please reopen immediately...

DonLakeFlyer, 2019-01-20:
The android problem is a dup of #6693. We can keep discussing that one there.