mavlink/qgroundcontrol #6716: PreFlight: Add support for momentary triggers

DonLakeFlyer, 2018-07-13:
For example it would be great to create a check list item which says blow across pitot tube and have it automatically go green when airspeed goes above a certain value. That is currently not possible. Current code requires checks to stay true, as opposed to being true for only a moment

Antiheavy, 2018-07-13:
A compass check could require the user to spin the vehicle around 360 during a 1 minute window or something, assuming the user is checking N-E-S-W directions for good calibration.

Antiheavy, 2018-07-13:
Servo/linkage checks could be presented. Buttons to check each actuator with + and - movement.

Antiheavy, 2018-07-13:
A button to “calibrate airspeed” would be a good one. This could call a “quick cal” function in the firmware. Airspeed readings after calibration could be required to be within a certain range of zero for X seconds.

DonLakeFlyer, 2018-08-02:
Too complex for last minute 3.4 add

SinigerVTL, 2018-08-05:
Good ideas. We use a release servo now on AUX3 which we use with set servo mission command. And there is no way we could test the servo on the ground like with the camera trigger. We are not sure if the set servo channel is total # or is it a AUX channel number in the mission set servo command. In one case we have to use ch 11 in another ch 3 in the set servo command. No way to test it...

The airspeed is another example for the preflight checks. Its reading becomes drifted in the first one minute or two then it settles that is with 5525 sensor. During this time the preflight checks passed with error. Then we have to restart the machine once more to get the correct reading and no preflight errors.

We've asked a couple of times if there is possibility to just zero the airspeed reading preflight with a soft button and not go through the long airspeed calibration procedure before each flight. BTW we do it manually this way with the change of parameter SENS_DPRES_OFF instead of airspeed calibration..

tubeme, 2018-08-05:
Good ideas. We use a release servo now on AUX3 which we use with set servo mission command. And there is no way we could test the servo on the ground like with the camera trigger. We are not sure if the set servo channel is total # or is it a AUX channel number in the mission set servo command. In one case we have to use ch 11 in another ch 3 in the set servo command. No way to test it...

The airspeed is another example for the preflight checks. Its reading becomes drifted in the first one minute or two then it settles that is with 5525 sensor. During this time the preflight checks passed with error. Then we have to restart the machine once more to get the correct reading and no preflight errors.

We've asked a couple of times if there is possibility to just zero the airspeed reading preflight with a soft button and not go through the long airspeed calibration procedure before each flight. BTW we do it manually this way with the change of parameter SENS_DPRES_OFF instead of airspeed calibration..

GaoGeolone, 2018-11-30:
@tubeme Have you test that function on a quadcopter?

Antiheavy, 2018-11-30:
> zero the airspeed reading preflight with a soft button and not go through the long airspeed calibration procedure before each flight.

this would be huge for us as well.  I'd say our most common pre-flight fail warning is airspeed drift related (usually temperature related).  Airspeed re-cal sounds like the easiest to implement too.

DonLakeFlyer, 2018-11-30:
> Airspeed re-cal sounds like the easiest to implement too.

Is there some different kind of firmware airspeed recal needed which doesn't exist yet? I don't quite what @tubeme is saying they do.

Antiheavy, 2018-11-30:
> Is there some different kind of firmware airspeed recal needed which doesn't exist yet?

Even pop-up a dialog from the pre-flight checklist that presented the currently implemented standard calibration would be helpful.  Would save several clicks and avoid having to leave the context of the Fly screen.

@dagar might have thoughts on this since I think he is the most recent author of the PX4 airspeed calibration? 

For future enhancement of this option, there has been discussion of a couple different yet-to-be-implemented airspeed calibration options in these firmware posts:
https://github.com/PX4/Firmware/issues/8463
https://github.com/PX4/Firmware/issues/9148

DonLakeFlyer, 2018-12-01:
@dagar Could the airspeed calibration be changed to not be text based status. If it used real mavlink messages for status it would be way easier for me to integrate it into a pre-flight button.

Antiheavy, 2020-02-15:
FYI @dakejahl not sure if anything you are working on applies to this old issue/request.