mavlink/qgroundcontrol #6720: GoTo command just hangs out onscreen after first use, never goes away

Antiheavy, 2018-07-15:
After using GoTo command for the first time, the icon just hangs around onscreen, even when the vehicle changes modes in and out of Hold mode or whatever.  Even hangs around for future missions.  Seems like it should disappear when the vehicle isn't going to there.

[image]

DonLakeFlyer, 2018-07-15:
Yeah, I can figure out some way/time to make that disappear.

Antiheavy, 2018-10-25:
> Do you think it would be ok for it to disappear as soon as you confirm it?

So then the vehicle would fly to a location on the map that had no marker indicating it?  I think it is best to show a location where the vehicle is heading to and holding at.  Is that information available from the vehicle?  If so, would displaying that as a marker make sense?

DonLakeFlyer, 2018-10-25:
> I think it is best to show a location where the vehicle is heading to and holding at. Is that information available from the vehicle?

Nope that is not available.

DonLakeFlyer, 2018-10-25:
> So then the vehicle would fly to a location on the map that had no marker indicating it?

yup

DonLakeFlyer, 2018-10-25:
The best I can do is get rid of it if the flight mode changes from Hold. But that only handles some of the case.

Antiheavy, 2018-10-26:
I think handling some is better than none here. That GoTo point is only valid in Hold mode so it seems fine to only display if either currently in Hold mode or if QGC is waiting for the confirmation slider (I.e about to enter Hold mode there).

DonLakeFlyer, 2018-10-26:
Better than nothing. Daniel has another Issue which covers a correct solution to this problem which requires firmware support.

DonLakeFlyer, 2018-12-18:
I looked at this a few days back. I hoped it would be easy but not so. Quite a large change. Still needs to happen though.