mavlink/qgroundcontrol #6723: Terrain bug, seeming Zero terrain height values for a couple WP after downloading off vehicle

Antiheavy, 2018-07-16:
Happened to notice this after downloading a mission from a vehicle (SITL).  WP 70 and 71 both show what looks like Zero MSL altitude which is incorrect for this location:

[image]

I tried switching from "Rel" to "AGL" for these two waypoints and they jumped down below the ground when "AGL" was used, confirming QGC had the wrong terrain value there (i forgot to screen shot that).

After adjusting WP 71, then the terrain updates to the correct terrain height value which confirms the wrong value was shown:
[image]

I believe I had a not-so-great internet connection because after uploading and downloading again I had a lot of yellow bars that persisted:
[image]
and terrain network errors in the console:
[image]

I cannot reproduce yet... been trying.

Antiheavy, 2018-07-16:
here is the mission plan file (attached zip folder).
[couple-WPs-missing-terrain.zip](https://github.com/mavlink/qgroundcontrol/files/2199267/couple-WPs-missing-terrain.zip)

DonLakeFlyer, 2018-08-02:
Or in other words the hash points to a tile but the validation says the coordinate is not in the tile due to precision error.

DonLakeFlyer, 2018-08-02:
#6755 Prevents bad surveys from being generated. But does not fix the underlying problem of the actual failure

DonLakeFlyer, 2018-08-02:
I believe the real fix is to change persistance of tile corners to doubles. But that is a file format change which is a little tricky.

DonLakeFlyer, 2018-08-03:
Even with this last pull there is still a bug here where this plan (
[WPs-missing-terrain-2018-08-02 copy.plan.txt](https://github.com/mavlink/qgroundcontrol/files/2258277/WPs-missing-terrain-2018-08-02.copy.plan.txt)
) no doesn't cause bad elevations to be loaded. But it still doesn't fully load elevation data.

DonLakeFlyer, 2018-08-04:
> Even with this last pull there is still a bug here where this plan

This problem is fixed now. Was caused by 0,0 coordinate flowing to terrain system from intermediate state during json load.

Lower lowel terrain query/tile code is now hardened to error. Next step is extra level of paranoia at upper level code checking for elevations below 0. Shouldn't happen due to lower level changes, but better safe than sorry.

DonLakeFlyer, 2018-08-16:
This should be all fixed up now

closed 2018-08-16 by DonLakeFlyer