mavlink/qgroundcontrol #6733: DJI CrystalSky Menu Bar Size

ve3otk, 2018-07-26:
Have installed latest (daily build) of QGroundControl (3.3.2-875) on DJI CrystalSky 550 (5.5") and 785 (7.85") and finding the top menu bar in the program too small to feasibly operate. The DJI tablets are running version 5.1.1 of Android. Have been told this is an issue with Android (older versions) and dealing properly with high resolution displays so some form of work around programming is required.

DJI CrystalSky 550 screen resolution is 1920x1080 (16:9 aspect ratio.) DJI CrystalSky 785 screen resolution is 2048x1536 (4:3 aspect ratio.)

There appears to be some scaling of the top menu bar based on the font size, but this does not work all that well on these units. Maybe some other way can be programmed to have a separate means of selecting how large the menu bars are that is not tied to the font size. 12 point font is perfectly readable on these units.

Hope some means can be found to make the menus easier to control on the DJI tablets, as they make a great Android tablet that works with our non-DJI applications and our USB based telemetry radios. Should also work with WiFi based telemetry. The DJI tablets have long battery life and high brightness displays - much better than any phone. (1000mcd and 2000mcd!)

Have attached a couple of screen captures of QGC's main page, one is with text font size at 12 and the second size at 24. From 12 to 20 points, the top title bar remains 62 pixels vertically on the CS785 unit. When font size is changed to 24 points, the top bar vertical size is still 62 pixels, but the top left corner button increases to about 76 x 76 pixels.

On the CS550, the transition from 62 x 62 pixels for the top left corner button changes to about 76 x 76 pixels just before font size of 20 points, and increases to about 96 x 96 at font size of 24 points. The top bar remains at 62 pixels vertically.

Also have an NVidia Shield tablet (1920x1200) running a newer version of Android and QGC displays great on it. The vertical size of the top bar is about 110 pixels. NVidia Shield tablet is out of production, does not have a high brightness display, and battery life is not much better/worse than a cell phone. Plus it has all the other stuff running in the background that is not present on the DJI tablets.

[image]
[image]

ve3otk, 2018-07-26:
Some screen captures from DJI CS550 at font size of 12, 20 and 24 points.
[image]
[image]
[image]

dogmaphobic, 2018-07-27:
Yes, I noticed that when I first got the CrystalSky (the larger one). It has a high density display but the system reports single density. That causes everything to be out of whack. I "fixed" it by running a "custom" build. The PR for it is here: https://github.com/mavlink/qgroundcontrol/pull/6678

I'm not sure how to come about a general fix.

dogmaphobic, 2018-08-09:
I worked around the issue but I was only able to test on the 7.85" version.

[image]

closed 2018-08-09 by dogmaphobic

ve3otk, 2018-08-16:
Looks good on the 5.5" unit, too. Thanks!

[screen]<https://user-images.githubusercontent.com/749243/43873107-3f3fe932-9b53-11e8-8f28-b6592b264f53.png>

[https://user-images.githubusercontent.com/749243/43873107-3f3fe932-9b53-11e8-8f28-b6592b264f53.png]

PR #6782 (2018-08-09): iOS build and mobile sizing
Bringing iOS build changes from Stable_V3.4.
Also dealing with (non reported) device pixel ratio on mobile devices.

Fixes #6733