mavlink/qgroundcontrol #6734: RTK base not detected in QGC on windows

ambikaverma, 2018-07-27:
using qgc ver 3.3.2

RTK base is not detected in qgc on windows using either the VCP drivers or the USB drivers.
Using the VCP drivers I can atleast see it in u-center software but USB drivers don't work at all.

In linux when running same version of QGC I did see the rtk base being detected.

rtk base - here+ base module
Any suggestions or fixes?
Thanks!

DonLakeFlyer, 2018-07-30:
@bkueng Should this be working with this RTK GPS?

bkueng, 2018-07-30:
Yes, but I'm not sure about Windows. I never tested it there.

DonLakeFlyer, 2018-07-31:
@ambikaverma Can you try the latest daily build and report results here?

ambikaverma, 2018-08-12:
Sorry for late reply. I'll try that and post the result later today.
Thanks!

ambikaverma, 2018-08-12:
I tried the daily build version of QGC but its still not able to recognize the base RTK.
It does see the COM port to which the base RTK is connected, but don't see it being recognized. I tried all the different data rates available, but no luck -
[image]

I tried mission planner to see if indeed something was wrong on ublox driver end, but mission planner is able to connect with the base RTK at the same COM port -
[image]
and I also get rtk float and fixed -
[image]

the same COM port is also accessible from u-center. So two separate software are able to connect with the HERE+ base rtk.

Anything else that I should try?

DonLakeFlyer, 2018-11-27:
@ambikaverma Any update on Beat's question?

closed 2026-10-04 by github-actions[bot]