mavlink/qgroundcontrol #6738: Secondary batters (bat2) not null value for volt/current and all on QGC Win and Android daily.

pompecukor, 2018-07-29:
I have a setup with dual current sensor (separate secondary battery and all), it works and all if is reported on Mission planner. However QGC shows -1 for relevant fields. This is a rearly used function, so I am guessing the mavlink data field was change on arduplane update this year and QGC did not follow up.
**Secondary issue** is there is no differnciating label on the secondary data, Like bat1 and bat2 volt are both labeled Volt. Should be **Volt** and **Volt2**, **Current and Current2** or something like that.

----
Testest on both Windows and Android, same result.
Daily builds.
The Cube (Pix2.1)
Aurduplane 3.9, 3.10

[image]

DonLakeFlyer, 2018-07-31:
Can you check the values coming in from the BATTERY_STATUS mavlink message using the Mavlink Inspector. The battery 2 values should some throught with id != 0. This is where the values come from.

pompecukor, 2018-07-31:
I will do next time I power on, but as mentioned, on Mission Planner it is working fine.
So the mavlink is coming in.

pompecukor, 2018-07-31:
And of course great job on you support. I admire you!!
I will post another issue which is to do with quadplane, take off and 0,0 lat and longitude. Take off should not have coordinates, so QGC ends up showing it in the middle of the word, which makes the mission look stupid. I will demonstrate.

But it is easy to replicate as you can just make any mission and add command Vtoltake off and add 0,0 coordinates. 

The reason why we cannot add a specific (like home) coordinate is that current VTOL mode on arduplane has a bug that is the VTOL take off has a coordinate it will start to fly in that directed even when taking off vertically, which is very bad. Sometime it can flip the plane as it is taking off. So some firmwares already "force" the command when being saved to have 0,0 coordinates. This then when synced on QGC looks bad.

kikislater, 2018-08-01:
@pompecukor it's same like here for 0,0 : https://github.com/mavlink/qgroundcontrol/issues/5272
Why not using planned home position ?

DonLakeFlyer, 2018-08-02:
> Mission Planner it is working fine.
> So the mavlink is coming in.

Sure. But the specific values in there might not be what I'm expecting.

DonLakeFlyer, 2018-08-03:
> I will post another issue which is to do with quadplane, take off and 0,0 lat and longitude. Take off should not have coordinates, so QGC ends up showing it in the middle of the word, which makes the mission look stupid. I will demonstrate.

Can you make sure to enter an issue with full details on this. I'm not quite understanding the problem.

DonLakeFlyer, 2018-08-03:
> I cannot find how to (during fixed wing missions) point to a place and othe map and do guided mode/fly to here. Also how to change altitude? Thx

You click in the map. But you first have to change from Auto mode.

DonLakeFlyer, 2018-08-03:
> Can you check the values coming in from the BATTERY_STATUS mavlink message using the Mavlink Inspector. The battery 2 values should come through with id != 0. This is where the values come from.

Realizing that checking this in mavlink inspector will not work. I will need a telemetry log to replay so I can figure this out.

closed 2026-10-06 by github-actions[bot]