mavlink/qgroundcontrol #6740: Confusing Tuning Slider for Plane Roll and Pitch Sensitivity

rollysing, 2018-07-30:
This is on the Daily Build:
QGroundControl Development HEAD:8aba44562 2018-07-2814:17:58-0400
But is also in:
QGroundControl Development HEAD:e3833aa5f 2018-07-13 11:43:25 -0700

Windows 10 Pro
QGroundControl Development HEAD:8aba44562 2018-07-2814:17:58-0400
Pixhawk 2.4.6 (Generic eBay)
PX4 Pro v1.8.0

So the confusing thing is for Plane Roll Sensitivity, how did 0.4s end up on the left side of 0.05s when 0.3s is on the other side. Same thing with Plane Pitch. why is 0.8 s is on the right of 0.95 s when 0.85 s is all the way to the left? What am I missing?

Personally, I prefer the original version with the Slider. Why not incorporate the numerical scale above or below the graduation tick marks with the slider?
Please advise. Thanks.

Param file:
[073018 Opterra 246 v1_8_0 X6R Generic Quad Delta Iniital.zip](https://github.com/mavlink/qgroundcontrol/files/2242922/073018.Opterra.246.v1_8_0.X6R.Generic.Quad.Delta.Iniital.zip)

DonLakeFlyer, 2018-07-31:
Is this a VTOL vehicle?

DonLakeFlyer, 2018-07-31:
Also can you include the origin picture of the problem?

DonLakeFlyer, 2018-07-31:
Nevermind I can reproduce.

closed 2018-07-31 by DonLakeFlyer