mavlink/qgroundcontrol #6766: update for new ArduPlane plane stable release

tridge, 2018-08-06:
Hi Don,
I've done a new stable release of plane for ArduPilot, version 3.9.0. A few issues I've noticed while testing with qgc 3.3.2:

- it doesn't seem to understand the new battery parameters, and shows this:
[image]

- it doesn't offer the ability to load firmware on the new ChibiOS boards, or load ChibiOS based firmware on the existing boards (such as Pixhawk1, CubeBlack, Pixracer etc).

- it doesn't seem to recognise the new ArduPilot bootloader at all. It uses the same protocol as the old PX4 bootloader, but uses different USB vendors and device IDs.

I suspect you are already well aware of these issues, but I thought an issue to track them could be useful. If you want more details on any of them let me know.

Cheers, Tridge

Kelly-Foster, 2018-08-06:
QGC Team, available to test any ChibiOS related PR's on multiple new F4 boards as well as Pixhawk Cube and PixFalcon. Thanks.

Kelly

DonLakeFlyer, 2018-08-06:
> it doesn't seem to understand the new battery parameters, and shows this:

Support for that is in daily builds and Stable 3.4 which will be out in a few days.

> it doesn't offer the ability to load firmware on the new ChibiOS boards, or load ChibiOS based firmware on the existing boards (such as Pixhawk1, CubeBlack, Pixracer etc).

Chibios/new bootloader support will go into 3.5 covered here: https://github.com/mavlink/qgroundcontrol/issues/6760. Will be available/implemented in daily builds once 3.4 is done and out the door.

DonLakeFlyer, 2018-08-06:
Thanks for testing @tridge

closed 2018-08-06 by DonLakeFlyer