mavlink/qgroundcontrol #6767: Set Servo Mission command does not seem to work.

tubeme, 2018-08-06:
Hello guys,

We are trying to use Set Servo Mission command without success.

[image]

Here we use channel 3 presuming the channel we use here are AUX channels.

This way on arming the servo goes to 1500 but then on set servo waypoint on ch 3 at 2000us nothing happens. The servo does not release.

Maybe we have to use absolute channel numbering so AUX ch 3 becomes ch 11

We use 1 month old master of QGC and 1.8 PX4 stable.

QGC is Windows version.

Any ideas?

DonLakeFlyer, 2018-08-06:
You need to ask this in a firmware forum. QGC is just relaying that command to firmware. I don't really know how the settings for that command work.

closed 2018-08-06 by DonLakeFlyer

GaoGeolone, 2018-11-30:
> Hello guys,
>
> We are trying to use Set Servo Mission command without success.
>
> [image]
>
> Here we use channel 3 presuming the channel we use here are AUX channels.
>
> This way on arming the servo goes to [...]

Have you figured out this issue？Please tell me ,thank you~