mavlink/qgroundcontrol #6768: Pop up message after compass or general calibration

simonegu, 2018-08-06:
Would be nice to have a pop up message at the end of calibration on PX4 to say to reboot or sending the reboot cmd. I usually restart my set up after calibration, but I notice that not everyone is maybe doing it, resulting in poor flight performance directly after the calibration. A pop up message would probably reduce the failure rate for novice people. Thoughts?

Antiheavy, 2018-08-06:
I have a similar/related feature request in the PX4 firmware repo. I feel like an important piece of this is that the vehicle should prevent arming after calibrations that “should require” a full reboot.  I would really like to see something happen in this area sooner rather than later ;-)
https://github.com/PX4/Firmware/issues/9030

DonLakeFlyer, 2018-08-06:
Easily done in QGC. I already do this for ArduPilot. Didn't know PX4 needed it as well.

DonLakeFlyer, 2018-08-06:
Ok, just to be clear which calibrations should specify reboot is needed: compass, gyro, accel, airspeed, level?

DonLakeFlyer, 2018-08-06:
Was going to try to get this into 3.4 before it goes out the door. But it is a bit more complicated than I thought. A little too much code change for 3.4 so going into daily.

DonLakeFlyer, 2018-08-06:
> Ok, just to be clear which calibrations should specify reboot is needed: compass, gyro, accel, airspeed, level?

From the related Issue it looks like compass only?

DonLakeFlyer, 2018-08-07:
Since this was only for compass cal I was able to get it into Stable 3.4

closed 2018-08-07 by DonLakeFlyer