mavlink/qgroundcontrol #6770: Weird KML

dogmaphobic, 2018-08-07:
[testpolygon.kml.txt](https://github.com/mavlink/qgroundcontrol/files/2264744/testpolygon.kml.txt)

If you open this KML (renamed to kml.txt so I could attach) and move the indicated vertex, things seem to go haywire.

[image]

[image]

DonLakeFlyer, 2018-08-07:
I'll take a look at some point. But not too concerned with really strange shapes working. More concerned that theoretically this case should work fine.

thomasgubler, 2018-08-29:
I had a quick look. The _polygonModel object that is used internally for the polygon in the transects generation seems to stay correct. (That can be seen from the screenshot: the white lines follow the correct polygon). I have yet to take a closer look at the _polygonPath object that (I assume) gets picked up in qml for display.

DonLakeFlyer, 2018-08-29:
Ah, interesting. They could be out of sync. The other thing is possibly a Qt bug where the UI is not updating correctly to _polygonPath changes. That wouldn't suprise me at all either.

thomasgubler, 2018-08-29:
PR with possible fix here: https://github.com/mavlink/qgroundcontrol/pull/6838

closed 2018-08-30 by dogmaphobic