mavlink/qgroundcontrol #6772: Proffesional Pix4D work not possible.

tubeme, 2018-08-07:
Hello guys,

Finally the 1.8 FW solved the problem with the triggers for us so now we have the triggers and could geotag the images from a survey with the QGC parser.

There are two problems for a serious Pix4D with this method. While there is Lat Lon and Alt information recorded in the image files there is no Pitch Roll and Yaw information.  

Also before I remember there was a text file generated from the parser with all the names of the images and their Lat Long Alt Pitch Roll and Yaw parameters.

I don't know if it is possible the parser to record the pitch roll and yaw in the image file, if not for a Pix4D precision mapping we need to generate the text file with this additional info on top of the lat Lon and Alt recorded in the current implementation.

Antiheavy, 2018-08-07:
Pix4D reads XMP metadata tags to find Yaw/Pitch/Roll (and estimated horizontal and vertical accuracy and some other things).  Someone would have to add XMP stamping code to the QGC Geotagger.

birchera, 2018-08-17:
As @Antiheavy mentioned attitude information can be added to XMP meta data, but there is no field in the EXIF format for it.

Implementing this will not be easy:
* Can be done via an xmp library, has to comply with licenses and cross platform requirements.
* Writing such a thing yourself would be pretty hard as well I imagine.

DonLakeFlyer, 2018-08-17:
Seems like I remember something about Pix4D understanding ArduPilot log files for stitching. I'm guessing this might be the reason why? So they can pull out this information from there.

closed 2026-10-06 by github-actions[bot]