mavlink/qgroundcontrol #6780: SSL error with Terrain on Android QGC 3.4

kikislater, 2018-08-08:
Ground station operating system : Android 7

    QGroundControl version and build : latest release from playstore : 3.4

    Autopilot board: not required

    Autopilot firmware : not required.

    Exact steps to reproduce the problem :
    Plan survey mission, add terrain

Same as here : https://github.com/mavlink/qgroundcontrol/issues/6681 but in Android. Could post log 

[image]

kikislater, 2018-08-09:
It works on my old Tablet with Android 4.4 but not in my phone with Android 7

DonLakeFlyer, 2018-09-28:
Possibly related to #6337

DonLakeFlyer, 2019-01-19:
This should be fixed now

closed 2019-01-19 by DonLakeFlyer