mavlink/qgroundcontrol #6781: Application crash when using "Follow Terrain" in a Survey

Antiheavy, 2018-08-08:
Repeatable with recent QGC daily builds.  Windows 10.

To reproduce:
1) delete all offline maps and default tile set
2) plan a mission with a large survey (e.g. >1 km across)
3) check ON "vehicle follows terrain"
4) start adding, deleting, and moving vertices and/or moving the whole pattern
5) sooner or later the application will crash.

This is repeatable on my machine with:
-  my old offline maps (old elevation data format?)
-  after deleting all offline maps.
- planning in an area that I have no offline maps available
- planning in an area that DOES have new offline maps, then dragging vericies to an area without offline maps, then Poof!

So far I am unable to repeat this after re-downloading new offline maps and planning inside that area.

Antiheavy, 2018-08-12:
Repeatable on multiple Windows 10 PCs and multiple internet connections.  Seems to work fine if there are pre-cached maps/elevation in the new format, but as soon as you try planning in areas with no pre-cached maps/elevation the application eventually crashes (drag vertices, etc).  Here is a download link to a crash dump file:
https://drive.google.com/file/d/1Q3-agWwkQ7NORUki9UNAE_opeAjS1q0J/view?usp=sharing

dogmaphobic, 2018-08-13:
> I think it has something to do with blowing away the previous query when you move the survey and create a new query set.

There is a similar issue when you switch in and out of offline maps quickly.

dogmaphobic, 2018-08-13:
https://github.com/mavlink/qgroundcontrol/issues/6685

DonLakeFlyer, 2018-08-13:
> There is a similar issue when you switch in and out of offline maps quickly.

Don't think that would be related this was fully within the Survey and upper level terrain query code. And just realizing as I type this Corridor Scan likely has the same problem. GOing there now...

DonLakeFlyer, 2018-08-13:
> And just realizing as I type this Corridor Scan likely has the same problem.

Nope wrong. Fix in is TransectStyleComplexItem which is common to both.

DonLakeFlyer, 2018-08-13:
Moving the correct fix to 3.5

Antiheavy, 2019-12-03:
> Moving the correct fix to 3.5

@DonLakeFlyer is this issue still valid?

DonLakeFlyer, 2019-12-10:
Nope, should have been ages ago

closed 2019-12-10 by Antiheavy