mavlink/qgroundcontrol #6783: QGC Master fails to upload mission to ArduPilot

lvale, 2018-08-09:
Hi @DonLakeFlyer 
[image]

defining just a simple mission, when trying to upload to vehicle, this comes up. 

Apparently part of the mission is uploaded.

DonLakeFlyer, 2018-08-09:
Firmware version and vehicle type?

lvale, 2018-08-09:
3.6 master quad and tradheli
on qgc master

DonLakeFlyer, 2018-08-09:
My guess is NaN on Heading to indicate not heading change. Can you quick try single waypoint with/without heading checked?

lvale, 2018-08-09:
you mean using parameter 4 ?
[image]
[tt0.plan.txt](https://github.com/mavlink/qgroundcontrol/files/2275928/tt0.plan.txt)

lvale, 2018-08-09:
i'm using the same mission that still fails

DonLakeFlyer, 2018-08-09:
Forgot that Heading field is not displayed for ArduPilot only PX4. Change Param4 to 0 (--.-- indicates NaN) and I bet it will work. NaN there is supposed to indicate don't change heading which is what param 4 is. Looks like something changes in ArduPilot to start rejecting this for some reason.

lvale, 2018-08-09:
Exactly. Changed the last WP to RTL

DonLakeFlyer, 2018-08-09:
Awesome, thanks. I'll fix. Defaults for params whould have been reset when you switched commands but they weren't. So you still have the NaN for heading in there from the original waypoint command.

lvale, 2018-08-09:
;) Thanks @DonLakeFlyer 
do you close this or do I?

DonLakeFlyer, 2018-08-10:
I'll close once I fix

Thyself17, 2019-03-29:
Sorry for stirring up the pot again, but I'm getting this on QGC 3.5.0 using Ardupilot Plane v3.9.6
All commands that include a loiter will fail the mission upload (including setting up a landing and a normal, infinite loiter).  

This can be repro'd by inserting a new waypoint, tapping the waypoint to change the type, selecting loiter from the dropdown, and selecting the loiter waypoint type.
At this point, you will not be able to upload a mission.  Param 4 shows as --.-- just as show in @lvale 's screenshots.

DonLakeFlyer, 2019-03-30:
@Thyself17 You problem is different than the original problem. This has to do with the LOITER_UNLIM command and param4 which specifies a heading. A NaN in this field means do not change heading. This is used for multi-rotor only so I'm not sure why ArduPlane is looking at that param at all since it is not relevant for fixedwing. I'm going to have to talk with Tridge to figure out why this matters to ArduPlane. You can workaround by going into Show all value and changing param4 to 0.

DonLakeFlyer, 2019-03-31:
In looking at ArduPilot source this is a generic problem with all vehicle types. It doesn't seem to validate param4 correctly for the LOITER commands. Although the mavlink spec on LOITER.param4/heading is not very clear as to NaN usage.

Thyself17, 2019-04-01:
@DonLakeFlyer Ah thanks for the update.
The workaround for changing param4 to 0 does work for loiters, but this is also preventing me from setting up a fixed wing autolanding.
Regardless, I see that this has been moved to the MavLink space, so thank you for escalating this in the right direction for me!  You guys are awesome.
I presume there will be a follow-on issue in ArduPilot to handle NaN?

DonLakeFlyer, 2019-04-05:
@Thyself17 Can you try tomorrow's daily build and see how it works?

Thyself17, 2019-04-05:
> @Thyself17 Can you try tomorrow's daily build and see how it works?

Will do.  Thanks for your tremendous efforts, @DonLakeFlyer .  I'll let you know how it goes.

DonLakeFlyer, 2019-04-05:
> but this is also preventing me from setting up a fixed wing autolanding.

Not quite sure what you mean by that can you explain? I only changed the code for adding a loiter manually.

Thyself17, 2019-04-05:
> but this is also preventing me from setting up a fixed wing autolanding.
> 
> Not quite sure what you mean by that can you explain? I only changed the code for adding a loiter manually.

I've attached a screenshot (and the mission plan .plan file renamed as a .txt) to hopefully help with the description.
To repro, I just setup a very basic "Fixed Wing Landing" pattern via mission plan.  Then click upload.  The upload fails with the "Param5 nan" message.

[image]

[Param5_MissionFail.txt](https://github.com/mavlink/qgroundcontrol/files/3049770/Param5_MissionFail.txt)

DonLakeFlyer, 2019-04-05:
Hmm, that's a different problem than the LOITER commands. It's caused by the stop camera commands. In that case as far as I can tell ArduPilot is just doing the wrong thing.

DonLakeFlyer, 2019-04-05:
Also I don't think ArduPilot supports those commands yet. Can you just turn those off?

Thyself17, 2019-04-05:
> Hmm, that's a different problem than the LOITER commands. It's caused by the stop camera commands. In that case as far as I can tell ArduPilot is just doing the wrong thing.

Gotcha.  I shouldn't have assumed they were directly related. Turning off the camera settings allows me to upload fixed wing landing patterns again.

And I'll make sure I get back to you with the results of the daily build tomorrow :)

DonLakeFlyer, 2019-04-06:
There is a bug that for ArduPilot these checkboxes should not be showing up at all.

DonLakeFlyer, 2019-04-06:
Just to detail out the issue. QGC was sending the wrong values for fixed wing missions and LOITER commands with respect to param4. I've fixed that. I've hacked ArduPilot multi-rotor mission upload to also default to 0 for param4 (instead of NaN) which to me seems incorrect but don't have time to argue. ArduPilot as far as I can tell does not yet support the heading field in loiter commands. What this means is that if they ever do they will suddenly all set of a heading of 0 which is basically wrong. PX4 multi-rotor missions continue to defailt to NaN to specify no heading change on loiter commands.

There are still problems here with respect to camera commands in Fixed Wing Landing pattern and ArduPilot. I'll get around to fixing that as well.

DonLakeFlyer, 2019-04-06:
Also realizing now that my hack for this has broken PX4 fixed wing since it will default to NaN for param4 when it should default to 0 for fixed wing for x-track exit.

Thyself17, 2019-04-06:
Well the good news is that with the daily build from yesterday I can successfully create loiter waypoints for Ardupilot Plane :)

DonLakeFlyer, 2019-04-16:
@Thyself17 Can you try tomorrows daily build again to make sure I haven't screwed anything up with latest change?

Thyself17, 2019-04-18:
@DonLakeFlyer  
Tested with Arduplane 3.9.6 using the daily build (4-16-2019).
Loiters still work fine.  No param issues on upload.  When looking at param values, param 4 is 0.00

Fixed wing landing pattern looks good too.  No camera options and the upload works fine

DonLakeFlyer, 2019-04-18:
@Thyself17 Awesome, thanks so much for veriyfing.

closed 2019-04-18 by DonLakeFlyer

Thyself17, 2019-04-18:
@DonLakeFlyer No problem.  If you ever need someone to verify changes to QGC on an arduplane build, throw me a ping.

PR #6789 (2018-08-13): Plan: Fix defaulting when you "Change Command"
* Param 1-4 are reset to 0
* Param 5-7 are reset to 0 depending on new command usage

Fixes #6783