mavlink/qgroundcontrol #6786: ArduPilot: Support to flash MindPx-v2 boards

lvale, 2018-08-10:
Hi @DonLakeFlyer 

As we've discussed, one of the supported boards on the new chibios releases are the MindPx

example for plane:
http://firmware.ardupilot.org/Plane/stable/mindpx-v2/

DonLakeFlyer, 2018-11-27:
@lvale With latest daily this is fixed now right?

lvale, 2018-11-29:
Recognises the board for regular setting up and flight.

Flashing still broken

[image]

Tangentially related. fmuv5 (CUAV v5) is flashing ok

lvale, 2018-11-29:
Sure

[firmwarefailed.txt](https://github.com/mavlink/qgroundcontrol/files/2627377/firmwarefailed.txt)

closed 2026-10-06 by github-actions[bot]