mavlink/qgroundcontrol #6791: Flight Data Replay Run at 6 Times Faster

rollysing, 2018-08-13:
The *.mavlink telemetry file is played at six times faster than normal speed. This is true with current stable v3.4.0 and tried it with v3.3.2, 3.3.0, 3.2.0. Had to go back to v3.0 for it to play in normal speed as default. Tried this in Windows 8.1 and 10 Pro PCs. Attached is the mavlink file and video. Telemetry was created with Pixhawk 1, PX4 v1.6.5 (I think).

[032817 Ranger EX QGC VTOL.zip](https://github.com/mavlink/qgroundcontrol/files/2284237/032817.Ranger.EX.QGC.VTOL.zip)

https://youtu.be/_w3TQu-kGQA

Please advise.

Thanks,

Antiheavy, 2018-08-13:
Replay is super convenient (logs already stored locally) and useful.  Ideas for future improvements in this old feature request: https://github.com/mavlink/qgroundcontrol/issues/5730

DonLakeFlyer, 2018-08-17:
Hmm, possibly a windows only thing (which would notbe unusual). I've been using replay a bunch on OSX, Linux without problems.

DonLakeFlyer, 2018-08-17:
I tested window using a log file generated from SITL flight and it worked fine. The problem seems to be specific to the attached log files. How are you getting these telemtry log files with a .mavlink extension? QGC hasn't used that extension for a while now.

rollysing, 2018-08-17:
That was generated Mar 28, 2017. As mentioned, I had to go back to QGC v3.0 so it replays at normal speed. So, you're saying that if I generate a new log with the new version that its replay would be normal? If that's the case, then I'll try it next time I fly. Thanks.