mavlink/qgroundcontrol #6794: Impossible to select the right airframe

Pauloledes, 2018-08-14:
Hi everyone,

I am currently facing an issue that I have never had before. The problem appears when I try to select an airframe for my Intel AERO Drone. In reality I have two problems.

The first one is that QGroundControl (v.3.4.0)  does not offer me the possibility to select the right airframe for my drone so I have to select one I think is the closest one of my real airframe. It is weird because I work with an other student who is able to select the right Airframe. The only difference between us is that he installed QGC on Windows whereas I installed it on Linux Ubuntu. Here is a picture of which airframes I can choose.
[image]

The second one is that QGC seems not to save the airframe I select. That is what my interface looks like after I selected a Generic Quadrotor X airframe.
[image]

It is a bit annoying since I can not do the other calibration properly because QGroundControl reacts like I never told it which airframe I am using.Then it does not know if it is connected to a Quadrotor or a plane for example.

Here are the full informations of the versions I am using : 

The firmware is Px4.

BIOS_VERSION = AERO-01.00.13
OS_VERSION = Poky Aero (Intel Aero Linux Distro) 1.6.1 (pyro)"
AIRMAP_VERSION = 0xc2
AeroFC firmware version = 1.6.5

QGroundControl 3.4.0

~Paul

DonLakeFlyer, 2018-08-16:
Most likely your cached airframe meta data is out of whack. Delete the PX4AirframeFactMetaData.xml in your ~/.config/QGroundControl.org directory. Boot QGC. Should show up.

DonLakeFlyer, 2018-08-16:
@LorenzMeier This seems to happen a bunch. Not sure that caching the airframe xml from flashing is such a great idea. That version seems to be screwed up sometimes, whereas the version in QGC is correct. But the cached version takes precedence so users get stuck.

Pauloledes, 2018-08-17:
Thank you for your help @DonLakeFlyer, now it works !

closed 2018-11-27 by DonLakeFlyer

zenetio, 2018-12-05:
I was getting the same issue but after many tests when using QGC v3.4.4 and pixhawk firmware v1.5.5
Then when I switched firmware to v1.8.2, then the issue was fixed.