mavlink/qgroundcontrol #6800: Corrupted altitude data in image geotagging

DanielePettenuzzo, 2018-08-15:
I tried geotagging images in qgc. I loaded my flight log and the images into qgc and everything seemed to work fine. When I open the image's metadata using exif I get the following metadata for the geotagging. Latitude and Longitude seem fine but the altitude data has an invalid format (ASCII and not Byte). 

```
GPS Tag Version                    |2.0.0.0
North or South Latit                  |N
Latitude                                     |47, 21, 1.731
East or West Longitu                |E
Longitude                                  | 8, 31, 20.537
Corrupt data
The data provided does not follow the specification.
ExifEntry: The tag 'GPSAltitudeRef' contains data of an invalid format ('ASCII', expected 'Byte').
```
Anyone knows where the issue could be?

DanielePettenuzzo, 2018-08-15:
FYI @dogmaphobic @LorenzMeier

DonLakeFlyer, 2018-08-16:
@birchera I don't know anything about the exif format. Does this ring a bell?

birchera, 2018-08-16:
No, I haven't ever seen that. Strange, because then lat and lon should have the same problem.

kikislater, 2018-08-17:
May be allowing to find solution, there is a note on exiftool documentation : 

> When adding GPS information to an image, it is important to set all of the following tags: GPSLatitude, GPSLatitudeRef, GPSLongitude, GPSLongitudeRef, and GPSAltitude and GPSAltitudeRef if the altitude is known

https://sno.phy.queensu.ca/~phil/exiftool/TagNames/GPS.html

So for latitude example, information should be written twice in exif with these GPSLatitude and GPSLatitudeRef. I had problem in the past to write exif in picture and it was solve applying exiftool rules !

DanielePettenuzzo, 2018-08-19:
Apparently the gpsAltRef has to be 0 or 1 butt the type of this value has to be uint8 instead of string. With this change I solved my issues: https://github.com/mavlink/qgroundcontrol/pull/6810
Now the exif editor doesn't give me the error anymore and when I feed the geotagged images to pix4d  the geotagging is used.
```
GPS Tag Version     |2.0.0.0
North or South Latit|N
Latitude            |47, 21, 0.594
East or West Longitu|E
Longitude           | 8, 31, 17.244
Altitude Reference  |Sea level
Altitude            |481.10
```

closed 2018-08-19 by LorenzMeier