mavlink/qgroundcontrol #6811: Feature Request: add support for dual GPS toolbar display

Antiheavy, 2018-08-20:
The autopilot firmware(s) support dual GPS (redundant GPS with blending).  We should be able to see the details of both GPS receivers in the toolbar and the info box when clicked.  There are lots of GUI options, but here is one suggestion:

[image]

Antiheavy, 2018-08-20:
Note:  according to @DonLakeFlyer the messages for this indicator come from MAVlink: GPS_RAW_INT

The 2nd GPS MAVlink info (GPS2_RAW) recently got added to PX4:
https://github.com/PX4/Firmware/pull/10265

moreba1, 2020-01-27:
@Antiheavy 
Can you share a log with dual gps ?

closed 2026-10-06 by github-actions[bot]