mavlink/qgroundcontrol #6813: Failsafe trigger from radio when throttle failsafe is removed

kikislater, 2018-08-21:
- Ground station operating system : Android 
- QGroundControl 3.4
- Autopilot board: Omnibus F4 pro v3
- Autopilot firmware : Arduplane 3.9.2 beta

Throttle failsafe disable in GCS, GCS failsafe enable (long events). Radio connected through SBUS with D8 mini Frsky. Power off radio trigger failsafe. 
As aircraft is in auto mission, it does not needed to have radio power on. Check and control were made through telemetry and Qgroundcontrol but power off radio trigger failsafe. Does not occur in Mission planner.

DonLakeFlyer, 2018-08-21:
The ground station is only in control of setting parameters. It has nothing to do with the actual failsafe trigerring. So this would seem to say that somehow the parameters are not being set correctly. Can you help verify the parameter settings themselves to see if something is wrong. If the parameter values are correct then the firmware folks are going to have to look at this.

DonLakeFlyer, 2018-11-27:
Other than settings up parameters QGC is not involved in failsafes. If you find a bug where QGC does not set the parameters correctly re-open this issue with details of that problem.

closed 2018-11-27 by DonLakeFlyer