mavlink/qgroundcontrol #6821: Joystick calibration is messing up more than it does good, increasing risk.

AndKe, 2018-08-24:
Current joystick calibration scales the joystick to ideal 1000us .. 1500...2000us range.
Then, it messes up all 12 channel config be it retracts, or any random feature or function.
This is not very nice, in fact, dangerous.

proper method would be:
calibrate joystick input , and scale it to the range of channels user already have
The calibration should read vehicle's parameters RC1_MIN   _TRIM  and _MAX , and scale the joystick data to that.
That would be MUCH safer, and did not interfere with everything on the other channels.

AndKe, 2018-08-24:
Yes, that is what I expected to believe, one of the joystick calibration buttons at the end changes text to something like "update firmware settings ?" or something like that - so I reviewed my dataflash logs - and found this (output from automated log parsing:)
RC1_MIN,1104  #updated from 1098
RC2_TRIM,1505  #updated from 1507
RC2_MIN,1111  #updated from 1105
RC5_TRIM,1500  #updated from 1170
RC5_MIN,1000  #updated from 1170
RC5_MAX,2000  #updated from 1810
RC6_TRIM,1500  #updated from 1913
RC6_MIN,1000  #updated from 1112
RC6_MAX,2000  #updated from 1913
RC7_TRIM,1500  #updated from 1512
RC7_MIN,1000  #updated from 1112
RC7_MAX,2000  #updated from 1913
RC8_TRIM,1500  #updated from 1113
RC8_MIN,1000  #updated from 1112
RC8_MAX,2000  #updated from 1913
RC9_MIN,1000  #updated from 900
RC10_MAX,2000  #updated from 1790
RC12_MIN,1000  #updated from 630
RC12_MAX,2000  #updated from 2350
RC12_REVERSED,0  #updated from 1
RC13_MIN,1000  #updated from 1112
RC13_MAX,2000  #updated from 1913
RC14_TRIM,1500  #updated from 1300
RC14_MAX,2000  #updated from 1790

Righjt now, I am perfectly confused how that happened unless QGC did it..

DonLakeFlyer, 2018-08-24:
Joystick cal doesn't touch parameters. It only writes to the .ini file. At least as far as I know. @jaxxzer ?

DonLakeFlyer, 2018-08-24:
I poked through the code. It's pretty much as I understood. There is no parameter futzing. Maybe the firmware folks do some mucking around with params when they get MANUAL_CONTROL messages. Don't know.

DonLakeFlyer, 2018-08-24:
Did you do an RC cal after you did a joystick cal maybe? Strangeness.

jaxxzer, 2018-08-24:
Is there some miscommunication here in nomenclature? @AndKe are you doing a usb joystick calibration, or an rc transmitter calibration?

AndKe, 2018-08-24:
@DonLakeFlyer   @jaxxzer No the RPAS in question is not new in any way. I did not do RC calibration , I just let the RC transmitter stay off, and connected Realflight interlink "radio" (Joystick) to the USB, and did the Joystick calibration. 
I'll reproduce it all in a few days.

AndKe, 2018-08-24:
OOPS . My bad, I just fetched the laptop & "interlink elite" and started to reproduce:
Discovered that the joystick calibration, and radio calibration is very much alike.
so @jaxxzer was right, I do know the difference very well, but simply chose the wrong one...

jaxxzer, 2018-08-25:
The RC calibration modifies parameters on the autopilot. The usb joystick calibration modifies settings local to QGC.

AndKe, 2018-08-25:
Yes, I am perfectly aware of that, only I thought I were doing joystick calibration and really selected RC calibration by mistake.
The mistake were built up by two things:
-Days before, I calibrated the interlink controller for SITL use, and remembered how the calibration looked.
-then , in field, when I wanted to calibrate, the interlink controller looks like a RC radio, so that helped me to choose wrong icon, the dialogs/design of the calibration is very similar, so I were "sure" I were doing joystick calibration,   - but - as you see me point out in the third post of this thread, I thought "writing to firmware" (or similar wording) were very strange...  (and I should have discovered my mistake then)