mavlink/qgroundcontrol #6823: Waypoint altitude not set correctly

sjd69, 2018-08-25:
QGroundControl release 3.4.1
Windows 10

QGroundControl does not properly set the altitudes of waypoints on 3.4.1. Each waypoints altitude is set to 0.0m. Toggling between relative and absolute altitudes for each waypoint fixes the problem.

Here's a sample mission:
[image]

Here's what is received by QGC:
[image]

DonLakeFlyer, 2018-08-25:
You changed the first waypoint from Takeoff to Waypoint right? I believe that is what causes it. Really bad bad all the same. Thanks.

DonLakeFlyer, 2018-08-25:
Hmm, it doesn't reroduce for me. Can you detail out the exact steps. And also save the .plan file which has the bad altitude in it even though the UI shows good altitudes?

DonLakeFlyer, 2018-08-25:
@antiheavy This theoretically doesn't affect you that much since you would always have the Takeoff command at the start. But there may be other cases which hit this which I haven't figured out yet. So to be safe, I think this fix means a new Stable for you.

Antiheavy, 2018-08-26:
I'm able to reproduce even with a Takeoff waypoint.

DonLakeFlyer, 2018-08-26:
@sjd69 Can you test the new version a bit. The installers are here:
* https://s3-us-west-2.amazonaws.com/qgroundcontrol/builds/Stable_V3.4/QGroundControl-installer.exe
* https://s3-us-west-2.amazonaws.com/qgroundcontrol/builds/Stable_V3.4/QGroundControl.AppImage
* https://s3-us-west-2.amazonaws.com/qgroundcontrol/builds/Stable_V3.4/QGroundControl.dmg

DonLakeFlyer, 2018-08-26:
@Antiheavy Can you try the new build as well?

DonLakeFlyer, 2018-08-26:
Sorry, old doesn't matter. Any plans. And double check altitudes and round trip values.

Antiheavy, 2018-08-26:
Something is wrong with "vehicle follows terrain" though.  the altitudes seems wrong, but I havent figured out exactly what is up yet - notice the "normal" waypoints 3,4,5 in this screenshot, the "follow terrain" waypoints that follow should not be that different in altitude:
[image]

Antiheavy, 2018-08-26:
and when you switch a normal waypoint to AGL mode it actually changes to the wrong altitude too.
notice slight increasing ground height under these waypoints:
[image]

then change the last one to AGL, you'd expect it to increase altitude slightly, but it actually goes to some lower altitude:
[image]

something is wrong.

Antiheavy, 2018-08-26:
i just noticed the "planned home position" seems to be at the wrong altitude too..  I can't seem to fix that...

DonLakeFlyer, 2018-08-26:
Ah, crap. I'm going to have to go back to drawing board on this one.

DonLakeFlyer, 2018-08-26:
@Antiheavy All the problems you are seeing are caused by the strange planned home position altitude. Is this Gazebo or a real vehicle connection. The actual height of the terrain based item should be the height difference from "P" to "T". If you eyeball it you can see that "6" looks about right. Same thing for the survey. So in reality this is working correctly.

If this is a real vehicle connection then it mainly shows the difference between the terrain data and your vehicle's actual altitude.

DonLakeFlyer, 2018-08-26:
When I test his on Gazebo it both cases are working fine for me.

Antiheavy, 2018-08-26:
> If this is a real vehicle connection then it mainly shows the difference between the terrain data and your vehicle's actual altitude.

Ahhh, this makes sense...  I was testing with a "real" vehicle (autopilot on a stick).  However, I'm indoors so my GPS solution and therefore actual altitude estimate was pretty bad.  If I unplug my GPS and re-do then things seem to work as expected.
[image]

The whole difference between terrain database and GPS reported altitude thing has been on my mind recently too, but that is a different issue.

DonLakeFlyer, 2018-08-26:
Yeah, it should clean itself up by round-tripping through QGC. Can you also load some correct real plans that you have have saved and just quick sanity check on load and round trip through vehicle.

DonLakeFlyer, 2018-08-27:
Thanks Todd. So far so good then.

DonLakeFlyer, 2018-08-27:
I"m going to mark 3.4.2 now and call it good.

PR #6825 (2018-08-25): SimpleMissionItem::setCoordinate was screwing up altitude/param7 syncronization
This could lead to missions being save with param7=0 but the ui shows a good altitude value.

Fixes #6823
closed 2018-08-28 by DonLakeFlyer