mavlink/qgroundcontrol #6824: Structure scan broken on ArduPilot

lvale, 2018-08-25:
Hi,

While trying to upload a simple structure scan to latest ArduPilot master copter one gets this

[image]

[crash_mission.plan.txt](https://github.com/mavlink/qgroundcontrol/files/2320268/crash_mission.plan.txt)

DonLakeFlyer, 2018-08-25:
This is because ArduPilot does not yet support the MAV_CMD_DO_SET_ROI_WPNEXT_OFFSET which is required to correctly point the vehicle or camera at the structure. I redesigned the ROI mavlink messages to support this (as well as to clean up thing that didn't work well) in coordination with PX4 and ArduPilot. PX4 implemented, ArduPilot did not. Maybe at some point it will get done. I didn't turn the feature itself off completely for ArduPilot with hopes that the user base would generate interest to get implemented.

closed 2018-08-25 by DonLakeFlyer

lvale, 2018-08-25:
Thanks @DonLakeFlyer