mavlink/qgroundcontrol #6831: Connections list do not list udp:14550 which is always open for incoming data

AndKe, 2018-08-27:
settings/connections show ttyUSB0  , ttyACM0  , but does not mention UDP:14550  
One would expect to see it there..

DonLakeFlyer, 2018-08-27:
Show it there so you can do what? If it's auto-connected you can't do anything with it.

AndKe, 2018-08-27:
Not seeing it there, feels like it's not enabled.  
Beside being more complete to have all connections listed, in an event like the one where QGC connected to SiK radio while I wanted it to use UDP, I had to think if , and test if QGC maybe did not see that UDP stream first.
Then I saw than adding a UDP connection, suggests port 14550 , which increases the feeling of this port not being listened in the first place.

The ultimate, super sweet solution would be to list UDP 14550 in that list too, and then be able to sort the list for desired priority. 

In short: as-is , any user that experiences something strange, or - does not get the UDP to work right away, will think it's not there.. misleading.

closed 2026-10-06 by github-actions[bot]