mavlink/qgroundcontrol #6832: Tuning screen reports missing parameters from beta version, firmware is stable version

dposea, 2018-08-27:
Firmware is Arducopter3.5.7, tuning screen reports missing PSC_ACCZ_*  parameters.  Those were released in 3.6 , changing the older ACCEL_Z_*  parameters.

DonLakeFlyer, 2018-08-27:
Can you check in vehicle summary page what version of firmware is being reported in there? It seems like QGC thinks the firmware is 3.6 or above.

dposea, 2018-08-28:
3.5.7

dposea, 2018-08-28:
I tried apm2 and it now has red values for two of the four standard tuning sliders. I dont know if that helps you find the problem. I'm going to use 3.6 for now and hope I dont have issues. This is for a bare bones hexa I'm using for fishing. No rangefinder, lidar, camera.

DonLakeFlyer, 2019-05-02:
This should all be cleaned up now.

MaEtUgR, 2020-02-25:
@DonLakeFlyer You might want to close this since there was no reaction.

closed 2020-02-25 by DonLakeFlyer