mavlink/qgroundcontrol #6839: Do not show "start mission" until there is a home lock

sanderux, 2018-08-30:
Do not show "start mission" until there is a home lock

DonLakeFlyer, 2018-08-30:
That starts to get tricky. You don't need a gps in all cases. Not sure how to handle all those cases where you can fly without home lock. For example optical flow I think can still fly missions?

Antiheavy, 2018-08-31:
+1 for this request. Maybe this should be tied to the vehicle status stuff @dagar has been talking about?

DonLakeFlyer, 2018-08-31:
> Maybe this should be tied to the vehicle status stuff @dagar has been talking about?

I think there may be too many variables for the ground station to know the right thing to do here. Without help from the firmware that is.

hamishwillee, 2018-08-31:
> How about a simple "ready to arm" or not boolean in mavlink?

@julianoes See this. Would help with SDK problem of knowing when ready to arm (since GPS state and other published health isn't sufficient due to preflight tests from EKF)

@dagar Would be good if we could ping for this.

sanderux, 2018-08-31:
Isnt altitude-rel an indicator? If that isnt set you are sure it cannot take off

DonLakeFlyer, 2018-08-31:
> How about a simple "ready to arm" or not boolean in mavlink?

Yup. That would work.

sanderux, 2018-08-31:
I think there is a difference between ready to arm and ready to start mission.
Arming is allowed without a home lock, mission is not.

Although i know that altitude-rel can be available even when the system isnt ready to start the mission, at minimum this will help in 90% of the cases.

Is a boolean was introduced perhaps we should add a bit more data in the state, one that would indicate what level of ready the vehicle is.

dagar, 2018-09-01:
Yes there can be a difference between ready to arm and ready to start mission, but all of this can be handled explicitly if we can work through the intended UX. 

I'd like to discourage doing this by proxy (eg relative alt). It's going to make the remaining arming problems even more mysterious.

DonLakeFlyer, 2018-09-01:
I also go back and forth on this sort of don't show this, or don't enable that type of ui. The reason is that you can end up with a system where users have no idea why things suddenly don't work or are invisible. Whereas if you try to Start Mission and it fails with a good error message then the user at least knows what is going on. Whereas if it simply doesn't show up then they are left in the dark. I'd lean towards folks like Todd who have lots of real users for feedback. And he's asking for it so I guess that a good indicator.

dagar, 2018-09-01:
I'd love to finally push forward on this. It's not necessarily a difficult problem, but requires coordination across Mavlink, QGC, and PX4. Should we organize a call?

dagar, 2018-09-01:
To be clear, what I'm interested in is removing all ambiguity (arming, mission, global position validity, etc) at the mavlink level.

From there how we use it is still up for discussion.

Antiheavy, 2018-09-01:
> For example optical flow I think can still fly missions?

is this true?  I don't know much about optical flow vehicles.

Back on @sanderux 's original topic of the Home position:  it seems to be that the Home position/altitude is fundamental to flying in auto modes.  Return mode, mission feasibility, etc are all highly dependent on a valid Home.

So whether there is a new MAVlink vehicle status message or not, the vehicle sending it's Home position to the ground station (I think it already does this?) would be a direct indication of at least this one piece of "ready to go" and "not ready, no home position" type status.

dagar, 2018-09-03:
> > For example optical flow I think can still fly missions?
> 
> is this true?  I don't know much about optical flow vehicles.

Potentially. It's why I keep bringing up position validity (local & global) and steer away worrying about GPS directly.

> it seems to be that the Home position/altitude is fundamental to flying in auto modes.

Actually not necessarily. Imagine you had a mission without any relative waypoints, RTL was set to use a planned mission landing, and the cylindrical geofence wasn't enabled. It's not currently implemented this way, but there's no reason why the state machine can't base these requirements on actual flight time requirements. I don't bring this up as a particularly interesting counter example, but to reenforce the idea that the vehicle can fairly easily make these decisions based on all kinds of different configuration and state. Then it gets even more complicated when we start talking about doing missions in local coordinates with things like optical flow.

So I don't think the ground should be trying to piece together what it thinks the vehicle needs for a specific situation (eg arming and starting a mission) when we could have the vehicle just broadcast it.

DonLakeFlyer, 2018-09-25:
> It's why I keep bringing up position validity (local & global) and steer away worrying about GPS directly.

For Start Mission should it just be home position available then as the criteria?

Antiheavy, 2020-02-15:
@sanderux there is now an option (in Master) to not show Start Mission until a preflight Checklist is passed. 

[image]

There is also an ability to use some auto-checking items for vehicle status like sensors and GPS health.   These possibly could be proxies for Home location, or at least they addressing your point of not showing Start Mission until the vehicle is "ready" (i.e. passed the whole checklist)
[image]

sanderux, 2020-02-16:
@Antiheavy thanks for the update. Unfortunately, as long as the PFCL is not editable (preferably model controlled) it is not something we are using / advising to our clients.

closed 2026-10-06 by github-actions[bot]