mavlink/qgroundcontrol #6846: Elevation data not available sent as NaN during Mission Upload

lvale, 2018-09-04:
If there is a problem with networking, or even no network available, defining a mission makes override the terrain height defined by the user and Mission Upload to vehicle sends a NaN which the flight stack does not support.

[image]
[image]

ChristopherOlson, 2018-09-04:
Would like to add that Android seems to work fine. The above screenshots are from my linux machine. The AirMap elevation data is cached fine with Android. The Linux version fails to get it. Also same problem noted on Windows version.

[image]

[image]

DonLakeFlyer, 2018-09-04:
What OS @lvale? Some versions of Android have problems with SSL libs. As do some versions of ArchLinux.

DonLakeFlyer, 2018-09-04:
That said I think there is an error path here where if the home position height query fails the NaN flows through to the top.

DonLakeFlyer, 2018-09-04:
@ChristopherOlson Are you using the latest version of QGC? Windows should be working fine.

lvale, 2018-09-04:
@DonLakeFlyer if i disable networking it happens on all OS's, But the biggest issue here is really the NaN coming up, and then that "kills" uploading missions

lvale, 2018-09-04:
latest build of QGC

ChristopherOlson, 2018-09-04:
@DonLakeFlyer I just referenced a related issue where it is happening on Windows. I don't actually use Windows, don't have a computer with Windows, have no need for a computer with Windows. On my Android tablet it works fine and fetches and caches the tiles for the AirMap elevation data within seconds. it is failing on two different linux machines even with internet connection.

Using the daily build on one and 3.4.2 on the other. Using 3.4.2 on the Android tablet.

DonLakeFlyer, 2018-09-04:
@ChristopherOlson What flavor/version of Linux?

ChristopherOlson, 2018-09-04:
Is there maybe a setting or library I'm missing? There shouldn't be any on the 18.04LTS machine because I build QGC daily build on that from source. But it has both Qt 5.9 (distro version), and 5.11.1 in it (to build QGC.

DonLakeFlyer, 2018-09-04:
To detail out the one case exactly:
* No network connection or SSL failures for query of terrain data
* Terrain height for Planned Home Position is **not** already in the terrain cache
* When you create a Plan the altitude for Planned Home Position will end up as NaN
* This will prevent you from uploading the Plan to the Vehicle

DonLakeFlyer, 2018-09-04:
@antiheavy FYI

DonLakeFlyer, 2018-09-04:
> Is there maybe a setting or library I'm missing?

It's a version problem with the SSL libs Qt wants. You should get console output from Qt about the failure.

DonLakeFlyer, 2018-09-04:
@ChristopherOlson Can you enter a seperate Issue for the problems you are having with terrain query failing?

ChristopherOlson, 2018-09-04:
@DonLakeFlyer I had a theory I had to check out. I'm on my Ubuntu 16.04LTS machine right now and I had QGC installed in ~/Documents/QGroundControl

I moved the installation to ~/QGroundControl, started it up and changed the setting for the path. Shut it down, restarted it, and voila! It got the terrain data instantly. I can't explain it. But when I get home I'm going to try this on the 18.04LTS machine and see if I get the same results.

DonLakeFlyer, 2018-09-04:
> I'm on my Ubuntu 16.04LTS machine right now and I had QGC installed in ~/Documents/QGroundControl

That's very strange. No idea why that would cause a problem.

DonLakeFlyer, 2018-09-04:
New daily builds with the fix should be up in about 1/2 hour.

DonLakeFlyer, 2018-09-05:
Just marked Stable 3.4.3 which has this fix in it.

closed 2018-09-05 by DonLakeFlyer