mavlink/qgroundcontrol #6848: Elevation data issues and waypoint planning

IronDonkey5, 2018-09-04:
This system is for posting bugs or feature requests ONLY. For questions about how to use or build QGC see: http://qgroundcontrol.com/#resources
Hi all, 
I am currently working on a custom quadcopter with a PixRacer R15 and APM 3.5
I am trying to create a waypoint auto flight around a field out back, and every time I attempt to upload the mission to the aircraft, I get the error "Unable to Save/Upload. Plan is waiting on terrain data from server for correct altitude values"
I have attempted to redownload the area of interest in offline maps but that did not seem to help. 

I am using a Dell laptop to create my mission and operate QGC as well as Mission Planner, I am using the normal/stable, newest build of QGC with APM 3.5 as stated above, 

Is there any fix for this or am I just overlooking something? 

Thank you

ChristopherOlson, 2018-09-04:
We're already looking at it with this issue https://github.com/mavlink/qgroundcontrol/issues/6846

IronDonkey5, 2018-09-04:
@ChristopherOlson  thank you, i didnt see that one til now.

DonLakeFlyer, 2018-09-04:
@IronDonkey5 Is the problem you are seeing #6846? Or it something different?

IronDonkey5, 2018-09-04:
@DonLakeFlyer  I honestly am not sure, the lingo in 6846 is a bit over my head. My problem is just as described in the topic above, anytime I attempt to upload a mission it gives me the "Unable to Save/Upload. Plan is waiting on terrain data from server for correct altitude values" error but I am unsure why. I am connected to the internet by hard wire, and I'm able to download maps for offline mode, although I haven't yet tried to upload a mission in offline mode.

DonLakeFlyer, 2018-09-04:
Does the Altitude of the Planned Home Position in Mission Start as shown in the image in #6846 have a "--.--" in it? If so, this is the same problem.

ChristopherOlson, 2018-09-04:
@DonLakeFlyer if it gets the terrain data that --.-- fills in with an altitude and then it loads a mission fine.

That "trick" that I did on my 16.04LTS machine did not work on 18.04LTS. It is failing to get the terrain data for some reason yet on 18.04. I tried both 3.4.2 and the daily build. I tried cleaning out the cache, deleting the .config files and started over. Still got the issue with no terrain data.

DonLakeFlyer, 2018-09-04:
Sorry, that question was for @IronDonkey5 to see if he is seeing the exact same problem.

IronDonkey5, 2018-09-05:
@DonLakeFlyer  I just tried to upload a waypoint mission again, and this time it randomly worked and uploaded. Honestly not sure what caused it to work this time.

DonLakeFlyer, 2018-09-05:
I'm closing this under the assumption you were running into the same problem

closed 2018-09-05 by DonLakeFlyer