mavlink/qgroundcontrol #6861: Unused variables in PX4FirmwarePlugin.cc

L4ncelot, 2018-09-10:
- Linux OS
- QGC master branch
- px4 firmware

Hi,

when I was going through the qgc I noticed something while setting flight mode. The problem seems to be [here](https://github.com/mavlink/qgroundcontrol/blob/master/src/FirmwarePlugin/PX4/PX4FirmwarePlugin.cc#L174-L184) . 
In this statement
```
            if (info.main_mode == px4_mode.main_mode && info.sub_mode == px4_mode.sub_mode) {
                flightMode = *info.name;
                found = true;
                break;
            }
```
variables `px4_mode.main_mode` and `px4_mode.sub_mode` are being tested but there's no assigning. Therefore it has always 0 value and won't return true ever. Shouldn't there be something like 
```
px4_mode.main_mode = base_mode;
px4_mode.sub_mode = custom_mode;
```
before the `foreach` cycle?

DonLakeFlyer, 2018-09-10:
If you look at the definition of px4_custom_mode you will see that it is a union.

closed 2018-09-10 by DonLakeFlyer

L4ncelot, 2018-09-11:
Sorry, I'm probably dumb... what's the point of going through `_flightModeInfoList` when `main_mode` and `sub_mode` are always 0? It won't recognize the combination of `base_mode` and `custom_mode` defined in `_flightModeInfoList` and always returns "Unknown flight mode".

L4ncelot, 2018-09-12:
I get it now. I just didn't understand the union struct. Thanks for help.

DonLakeFlyer, 2018-09-25:
No problem