mavlink/qgroundcontrol #6864: QGC not starting mission for multiple vehicles

exp131, 2018-09-10:
Hello,
I'm connected several PX4 drones to QGC by TCP. All connections are goes fine but when switch QGC to multi-vehicle mode and click on "Start mission" for all vehicles nothing happens. If start mission for each particular vehicle one by one, everything working fine also, but not simultaneously. Is this a bug or this feature not implemented yet?

OS: MacOS 13.10.5
QGC Version: 3.4.2
AP: Pixhawk 2.4.8 + PX4 firmware (lastest stable)

Regards,
Vitaly.

DonLakeFlyer, 2018-09-10:
This should be working. Likely a bug. Will take me a while to look at it since I'm heading out of town. Likely a simple fix. Just to be safe ping me again here sometime after the 24th so I remember to look at this.

DonLakeFlyer, 2018-09-25:
Fix is available in new daily builds. Available in Stable bulids when we release the next point version.

hughhugh, 2019-08-09:
Hi, I have connected two vehicles with the QGC, I can manually control the vehicle one by one. Then how could I start the missions of the two vehicles at the same time.  ‘SYSID_THISMAV’ parameter for both UAVs are different.
Thank you.

DonLakeFlyer, 2019-08-13:
Switch to Multi-Vehicle mode from the Fly view. Click Start Mission

huichoger, 2020-03-04:
Hi, I have connected 8 drones via wifi, all of them have different ports and ID, if I start the mission in the Multi-vehicle mode they all start with some seconds of difference, is there a way for all of them to start at the same or to reduce this time? 
thanks!

DonLakeFlyer, 2020-03-04:
> is there a way for all of them to start at the same or to reduce this time?

That is not possible