mavlink/qgroundcontrol #6867: QGC 3.4.3: RTSP Streaming does not work. NO VIDEO

tubeme, 2018-09-11:
We use the Windows version of master from 05-07-2018 and the current qgc 3.4.3.

In both cases the RTSP stream is not visible in the video window. It says "Waiting for Video".

The stream is working in all other players I have.

dogmaphobic, 2018-09-11:
If you don't have it already, download the ffmpeg tools and run:

`ffmpeg rtsp://<your_rtsp_uri>`

That will tell us what kind of stream it is. What is the source any way?

dogmaphobic, 2018-09-11:
Forgot to paste the link: https://www.ffmpeg.org/download.html

tubeme, 2018-09-12:
It is a h264 encoder box. I have decent control over the stream parameters:

[image]

[image]

tubeme, 2018-09-13:
Here is what ffmpeg brings:

Input #0, rtsp, from 'rtsp://192.168.40.25/0':
  Duration: N/A, start: 0.000000, bitrate: N/A
    Stream #0:0: Video: h264 (High), yuv420p(progressive), 1920x1080, 25 fps, 60 tbr, 90k tbn, 50 tbc
    Stream #0:1: Audio: aac (LC), 44100 Hz, stereo, fltp

dogmaphobic, 2018-09-13:
I'm not a GStreamer expert. In fact, this is an area we desperately need help with. From the encoder settings panel, do you know what _Compatible with FFMPEG_ means? What are the other options? Also, the resolution is too high for the given bitrate. I would stick to 720P at 2,048bps. _High Profile_ is not at all appropriate for over the air streaming and certainly not for such low bit rate. See if it offers _Extended_ or else _Baseline_. The other values are pretty much Greek to me.

AlexKlimaj, 2018-09-19:
I am also having issues getting an RTSP stream to work in QGC. I can get it working on VLC no problem.

dogmaphobic, 2020-02-11:
Ancient history

closed 2020-02-11 by dogmaphobic